shivasiddharth/GassistPi #274: Raspberry not showing in google home app devices

blackwiz4rd, 2018-03-18:
Hi,
I have seen from your youtube video that you the raspberry pi lists on your home app devices. In my case it doesn't. Do you know how this problem could be solved? Thanks a lot ;)

aandroide, 2018-03-18:
before opening the app assistant on your phone, you need to set the English language for the phone after open the assistant app and your devices will appear

blackwiz4rd, 2018-03-18:
Hi,
I have the language set in English US, does it have to be UK?

shivasiddharth, 2018-03-18:
1. You need to have an android device, not apple.
2. Android device should be running Android 7 and above.
3. Device language should be set to English US.
4. Pi gets listed in Assistant App and not in Home App.

blackwiz4rd, 2018-03-18:
Sorry for me being a noob. I had download 'Google home app' and not 'Google assistant app'

closed 2018-03-18 by blackwiz4rd

kodxana, 2018-05-21:
Do you know any way to acces assistant app from pc?