shivasiddharth/GassistPi #276: bug say in pushbutton

aandroide, 2018-03-18:
hello, in pushbutton, I noticed that in any function integrate the say function ("text") every time when I press the button to give the commands, he does not listen, he starts listening when I press the button for the second time.

aandroide, 2018-03-25:
I do not think it's a problem of say why in hotword everything works fine, it does so only in pushbutton. it would seem that it does not listen to the first push of the button after the stream commands, by pressing a second time the button returns to work

shivasiddharth, 2018-03-28:
I am sorry but unable to understand as to what the issue here is. Can you try and explain it a bit more.

aandroide, 2018-03-28:
simply if I say stream "song title" and then stubborn the song with the button, when I press the pushbutton to give a new command, the first time it flashes it sounds but does not listen to me, as soon as it stops flashing, I'll reset the button and go back to work correctly. I noticed that this defect only occurs in functions that integrate the say function ("text")

aandroide, 2018-05-25:
the problem disappeared with the update.
I close

closed 2018-05-25 by aandroide