shivasiddharth/GassistPi #279: youtube wont play when I try to stream a song.

nickallen176, 2018-03-21:
when I say hey google, stream"whatever song" it doesnt play. I waited for around 30 seconds and nothing played. It plays radio 2 fine but wont play youtube.

nickallen176, 2018-03-21:
I think it restarts when I stream from YouTube so I don't think thats the issue.

shivasiddharth, 2018-03-21:
Try the YouTube commmand, when it crashes, a log file will be generated in the /tmp folder attach that here. 

I am assuming that you went through the readme doc and setup the credentials.

MrPePperNL, 2018-03-24:
Hi

I might have the same problem. Rpi 3 with Raspbian Stretch.
Command i say to gassistpi: 'hey google, stream we will rock you' , errors in log and a breif message google assistant is fetching youtube. after that gassistpi is automatically restarted.

Errors in log:

2018-03-24 18:15:27,192 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com
2018-03-24 18:15:27,374 DEBUG requests.packages.urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-03-24 18:15:27,422 DEBUG requests.packages.urllib3.connectionpool https://translate.google.com:443 "GET /translate_a/single?tl=en&client=t&hl=en&ssel=0&q=Getting+youtube+link&dt=at&dt=bd&dt=ex&dt=ld&dt=md&dt=qca&dt=rw&dt=rm&dt=ss&dt=t&sl=auto&tsel=0&tk=102855.516774&oe=UTF-8&otf=1&ie=UTF-8 HTTP/1.1" 200 None
2018-03-24 18:15:27,432 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com
2018-03-24 18:15:27,552 DEBUG requests.packages.urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-03-24 18:15:27,577 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com
2018-03-24 18:15:27,685 DEBUG requests.packages.urllib3.connectionpool https://translate.google.com:443 "GET /translate_tts?tl=en&client=tw-ob&total=1&q=Getting+youtube+link&tk=102855.516774&idx=0&ie=UTF-8&ttsspeed=1&textlen=20 HTTP/1.1" 200 6720
2018-03-24 18:15:29,465 WARNING googleapiclient.discovery_cache file_cache is unavailable when using oauth2client >= 4.0.0
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0
2018-03-24 18:15:29,470 INFO googleapiclient.discovery URL being requested: GET https://www.googleapis.com/discovery/v1/apis/Youtube/v3/rest
2018-03-24 18:15:29,608 WARNING googleapiclient.discovery_cache file_cache is unavailable when using oauth2client >= 4.0.0
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0
2018-03-24 18:15:29,611 INFO googleapiclient.discovery URL being requested: GET https://Youtube.googleapis.com/$discovery/rest?version=v3
2018-03-24 18:15:29,912 ERROR __main__ name: Youtube  version: v3
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 420, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 296, in main
    YouTube_No_Autoplay(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1189, in YouTube_No_Autoplay
    fullurl,urlid=youtube_search(track)
  File "/home/pi/GassistPi/src/actions.py", line 254, in youtube_search
    developerKey=DEVELOPER_KEY)
  File "/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py", line 133, in positional_wrapper
    return wrapped(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery.py", line 240, in build
    "name: %s  version: %s" % (serviceName, version))
googleapiclient.errors.UnknownApiNameOrVersion: name: Youtube  version: v3

PS: Code tags are not working for me ..

shivasiddharth, 2018-03-24:
@MrPePperNL , You have got the credentials for wrong API/have added a wrong API in the cloud console.  And also, if you are behind a firewall/VPN sometimes, google authentication tends to fail. Can you please double check the API that you have added to the cloud console.

MrPePperNL, 2018-03-24:
I double checked, api added in cloud console is : YouTube Data API v3 with the same key as added to the /home/pi/Gassist/src/action.py section 
#YouTube API Constants
DEVELOPER_KEY = 'xxxxxxxxxxxxx'
YOUTUBE_API_SERVICE_NAME = 'youtube'
YOUTUBE_API_VERSION = 'v3'

shivasiddharth, 2018-03-24:
You have changed it in the wrong place. The key should be entered in the line above that reads:
API Key for YouTube and KS Search Engine
google_cloud_api_key='ENTER-YOUR-GOOGLE-CLOUD-API-KEY-HERE'

But that should not be an issue. Just saying. 
If the API key correct, then something in your network is blocking the authentication.

nickallen176, 2018-03-24:
I've added the api key and after a reboot I don't hear the startup sound,, I'm guessing it's because I've added the key to the YouTube api.

nickallen176, 2018-03-24:
I copied the api key and not the dev key so, I need to find that. It might work again.

nickallen176, 2018-03-24:
I'm on Google cloud but where can I find the api KEY?

shivasiddharth, 2018-03-24:
@nickallen176 , You can find all the requisite details here: https://github.com/shivasiddharth/GassistPi#music-streaming-from-youtube

nickallen176, 2018-03-24:
I realize that I added the api key to dev key for YouTube api constants and not Api for YouTube and KS search engine, so I changed YouTube api to the default (google_cloud_api_key) and add the key to api key for YouTube and KS search engine, but I don't hear the startup sound after a reboot.

shivasiddharth, 2018-03-24:
Open a terminal and execute:
`sudo systemctl status gassistpi-ok-google.service`
Paste here the contents of the terminal and also attach the log file here. Log file can be found in /tmp folder. Please dont paste the log file contents, attach them.

nickallen176, 2018-03-25:
I got it working again by replacing the old action.py with a new one then pasted the api key onto api for YouTube and KS search engine, so I hope this works. Also sometimes I get "hmm, something went wrong, try again in a few seconds or there was a glitch, try again in a few seconds"  error. Do you know how to make it stop doing that?

nickallen176, 2018-03-25:
The next day it's works but after a few hours I get those errors like "hmm, something went wrong, try again in a few seconds" or "there was a glitch, try again in a few seconds. "

MrPePperNL, 2018-03-25:
First of all, sorry for yjacking this issue ;)

I also got it working. There was no line google_cloud_api_key in my actions.py. It somehow got deleted even from my original actions.py. I've added it , restarted gassistpi and get a nice stream of music from youtube.

nickallen176, 2018-03-25:
i tried the youtube cammand and heres the log: 2018-03-25 19:10:27,135 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-03-25 19:10:27,149 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-03-25 19:10:27,508 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-03-25 19:10:27,561 INFO root Connecting to embeddedassistant.googleapis.com
2018-03-25 19:10:27,583 WARNING root Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
2018-03-25 19:10:27,584 INFO root Registering device
2018-03-25 19:10:27,588 DEBUG requests.packages.urllib3.util.retry Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2018-03-25 19:10:27,601 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): embeddedassistant.googleapis.com
2018-03-25 19:10:28,113 DEBUG requests.packages.urllib3.connectionpool https://embeddedassistant.googleapis.com:443 "POST /v1alpha2/projects/project-id/devices HTTP/1.1" 403 None
2018-03-25 19:10:28,123 ERROR root Failed to register device: {
  "error": {
    "code": 403,
    "message": "Could not create the device instance. Account doesn't have permission to perform this operation for project project-id. See https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device",
    "status": "PERMISSION_DENIED"
  }
}