shivasiddharth/GassistPi #283: headless auto start

codelock13, 2018-03-25:
hey i followed readme for headless auto start but after reboot i would not auto start when i do it manually it crashes every now and then i have just tried to stream music from youtube.also it plays wrong song.
when i told faded it played sing me to sleep.i am using aiy kit.
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/1845247/GassistPi.log)

shivasiddharth, 2018-03-25:
Can you check your .asoundrc and asound.conf files. Both should have same content. Raspbian tends to modify the .asoundrc file. If you have entered the required id in the systemd file then it should auto start without issue. Can you double check that. the id should be within single quotes.

codelock13, 2018-03-25:
Do you mean the model id in OK google service should be in single quotes
and I should add the service again?

shivasiddharth, 2018-03-25:
open a terminal and run:
```
sudo systemctl stop gassistpi-ok-google.service
sudo systemctl disable gassistpi-ok-google.service
```
This will remove the systemd files from boot. Now add the id within single quotes, save and run the service installer script and then:
```
sudo systemctl enable gassistpi-ok-google.service
sudo systemctl start gassistpi-ok-google.service
```

codelock13, 2018-03-25:
Issue solved thanks for your support

shivasiddharth, 2018-03-25:
Closing this thread.

closed 2018-03-25 by shivasiddharth