shivasiddharth/GassistPi #285: YouTube streaming

chathurangawijetunge, 2018-03-27:
More then 50% times assistant reboots after saying "getting youtube link"
Why is That?

patricktelling, 2018-03-27:
GassistPi saves a log file in the /tmp directory. You can use it to figure out what is causing the crash, or post your log file here.

chathurangawijetunge, 2018-03-28:
Why is this?

shivasiddharth, 2018-03-28:
This is not what I would expect to see in case of an error. Looks like you have restarted the Pi. The logs will be deleted on shutdown/restart. If you want to post an error log, the log file should be added here before shutting down or restarting the Pi. 

And please attach the log, don't paste it.

chathurangawijetunge, 2018-03-28:
If #287 is "Zombie Process" in linux.
Then this is also one of those zombies too...

shivasiddharth, 2018-03-28:
@chathurangawijetunge ,  If you are unaware of something and if someone gives a pointer to learn should either try and understand or request clarification instead of making fun. This kind of behaviour is unappreciated.  As a result, i am closing all your existing issues.

closed 2018-03-28 by shivasiddharth