shivasiddharth/GassistPi #286: If "chromecast" is spoken, application terminates

lottaboost, 2018-03-27:
If you say "ok google", then say anything with the work chromecast, the application exits.

shivasiddharth, 2018-03-27:
I have assigned a custom action with chromecast keyword. Open the main.py and scroll over to line 381 that says `if 'chromecast'.lower() in str(usrcmd).lower():` and comment the lines from 381 through 386.

lottaboost, 2018-03-27:
Nevermind, I had to set my chromecast ip address in the actions.py file.  I would have figured it out faster, but the chromecast did respond correctly even without injecting it's ip address into actions.py.

shivasiddharth, 2018-03-27:
The current SDK supports chromecasting out of the box, you dont need the custom action and hence, comment out those parts.

lottaboost, 2018-03-27:
Yeah, I commented out those lines.  And it still worked, I guess that custom action can be removed now.  thanks

lottaboost, 2018-03-27:
Now, hopefully they will update their SDK so pandora and music stuff can be used on device itself :)

shivasiddharth, 2018-03-27:
Issue resolved, closing this thread.

closed 2018-03-27 by shivasiddharth