shivasiddharth/GassistPi #287: Assistant re-starting

chathurangawijetunge, 2018-03-28:
try saying "ok google" and dont say anything..... try to do this for 2,3 times it restarts.
the erroe log is a follows 

----
2018-03-28 10:01:13,055 ERROR __main__ psutil.NoSuchProcess no process found with pid 3008
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/psutil/_pslinux.py", line 1390, in wrapper
    return fun(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/psutil/_pslinux.py", line 1572, in create_time
    values = self._parse_stat_file()
  File "/home/pi/env/lib/python3.5/site-packages/psutil/_common.py", line 337, in wrapper
    return fun(self)
  File "/home/pi/env/lib/python3.5/site-packages/psutil/_pslinux.py", line 1429, in _parse_stat_file
    with open_binary("%s/%s/stat" % (self._procfs_path, self.pid)) as f:
  File "/home/pi/env/lib/python3.5/site-packages/psutil/_pslinux.py", line 187, in open_binary
    return open(fname, "rb", **kwargs)
FileNotFoundError: [Errno 2] No such file or directory: '/proc/3008/stat'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/psutil/__init__.py", line 368, in _init
    self.create_time()
  File "/home/pi/env/lib/python3.5/site-packages/psutil/__init__.py", line 699, in create_time
    self._create_time = self._proc.create_time()
  File "/home/pi/env/lib/python3.5/site-packages/psutil/_pslinux.py", line 1401, in wrapper
    raise NoSuchProcess(self.pid, self._name)
psutil._exceptions.NoSuchProcess: psutil.NoSuchProcess process no longer exists (pid=3008)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 479, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 288, in main
    process_event(event, assistant.device_id)
  File "/home/pi/GassistPi/src/main.py", line 171, in process_event
    if ismpvplaying():
  File "/home/pi/GassistPi/src/main.py", line 94, in ismpvplaying
    p=psutil.Process(pid)
  File "/home/pi/env/lib/python3.5/site-packages/psutil/__init__.py", line 341, in __init__
    self._init(pid)
  File "/home/pi/env/lib/python3.5/site-packages/psutil/__init__.py", line 381, in _init
    raise NoSuchProcess(pid, None, msg)
psutil._exceptions.NoSuchProcess: psutil.NoSuchProcess no process found with pid 3008

shivasiddharth, 2018-03-28:
This is because of something called a "Zombie Process" in linux. Please do a quick Google search if you need more info. There is nothing much I can do than adding an exception.

closed 2018-03-28 by shivasiddharth

chathurangawijetunge, 2018-03-28:
So there is no way of stopping this....