shivasiddharth/GassistPi #291: repeating start sound

zacharywendholt, 2018-03-29:
After starting the assistant without shutting down it worked perfectly. After restart it now continuously repeats the starting/test noise. Any help fixing this problem would be appreciated

zacharywendholt, 2018-03-29:
to clarify. it is the same sound when the tune plays and the says "hi". after that the google assistant seems to still be working, but this sound continues to play every couple of seconds. I'm not sure if this will help, but it only happened when the device was connected to the internet also

shivasiddharth, 2018-03-29:
I suppose you are using the AIY HAT? And can you also confirm whether the systemd file was edited and device model id added?

shivasiddharth, 2018-03-29:
Please go through the readme. I have covered everything in detail in that.

zacharywendholt, 2018-03-29:
Thanks for the help!

kennethvkennedy, 2018-04-08:
I'd followed all the steps but faced the same issue as you did @zacharywendholt .

I fixed it by running the following commands:

`sudo systemctl disable gassistpi-push-button.service`
`reboot`

shivasiddharth, 2018-05-24:
Closing this for inactivity.

closed 2018-05-24 by shivasiddharth