shivasiddharth/GassistPi #292: Problem with registering custom device actions (with Google actions)

KaeYan, 2018-03-29:
Hi Please Help!

I want to create my own custom command to control my robot with the device actions, however, when I followed the steps in this link https://developers.google.com/assistant/sdk/guides/library/python/extend/custom-actions

The actions does not work, the assistant does not know how to help with the command, for this example, it is "Ok Google, blink the light 5 times"

Would really appreciate any helps!

shivasiddharth, 2018-03-29:
This project is an update behind the upstream SDK. The custom actions work only from 0.4.4. This project is on 0.4.3

KaeYan, 2018-04-01:
Hi Shiva,

Thanks for the reply! I will check the version of the google assistant version and get back to you.

KaeYan, 2018-04-03:
Hi Shiva,

Appreciate your reply!

I re-registered and deploy the action package for the blinking light tutorials, it works well. However, now I am trying to create my own device package. I went through gactions update and gactions test, however I think due to the schema error, I cannot trigger my custom command.

Can you/anyone help me to check if there is any error in the schema? Basically, I want to trigger a simple actions, the command is "I want to go to (goal) now", and I want to capture the (goal) as the parameter.
[actions.zip](https://github.com/shivasiddharth/GassistPi/files/1870755/actions.zip)

Thanks!

shivasiddharth, 2018-05-24:
The latest update to SDK 0.5.0 should solve this issue.

closed 2018-05-24 by shivasiddharth