shivasiddharth/GassistPi #295: OSMC branch does not support speaker-test, sound does not play out of speaker, raspberrypi-kernel package is not found

sokkycakes, 2018-03-29:
Basically what's in the title, there's a bunch of things that make doing this OSMC install harder. I'm using OSMC because i can't get NOOBS/dual booting to work for me and i want to keep using OSMC on my Pi B+.
speaker-test isn't installed and must be installed with alsa-utils. When testing it after doing the installation scripts, no sound comes out. The raspberrypi-kernel package install is impossible due to it not being in OSMC's repo. These issues must be fixed for this branch to remain relevant or usable at all.

shivasiddharth, 2018-03-30:
If you are running OSMC, you dont need to install raspberrypi-kernel. For the audio to work, the audio device must be selected within OSMC. I created it as a test branch and i dont think anyone used it/provided feedback and hence i left it as such and it is trailing a number of commits when compared to master.

devyez, 2018-04-09:
Hello, I'm trying to install following your GassistPi master branch on a OSMC running on a Raspberry Pi3. Ran into couple of issues, was able to tackle few by installing git and pip separately, but I'm stuck now. Any help is appreciated.

Below is what I have done so far, so that you get a better picture.

- Flashed the latest version of OSMC using OSMC installer on a blank, formatted SD card.
- Setup OSMC successfully on RPi3 and made necessary changes in port number, credentials. I'm able to use kodi and control it via mobile app.
- Connected to RPi3 via SSH and tried the installation steps in the master branch
- Installed git separately, cloned the repo. Ignored the kernel update command and sound tests.
- Copied JSON file to RPi3 and was able to run the gassist-installer-pi3.sh after installing pip using "sudo apt-get install python3-pip"

Now, I'm stuck after "Collecting RPi.GPIO" step. Below is the excerpt from the log:

creating build/temp.linux-armv7l-3.5/source
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-haFsHd/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/osmc/env/include -I/usr/include/python3.5m -c source/py_gpio.c -o build/temp.linux-armv7l-3.5/source/py_gpio.o
    unable to execute 'arm-linux-gnueabihf-gcc': No such file or directory
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    
    ----------------------------------------

Command "/home/osmc/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-sna23u77/RPi.GPIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-5nuglbnv-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/osmc/env/include/site/python3.5/RPi.GPIO" failed with error code 1 in /tmp/pip-build-sna23u77/RPi.GPIO/

[image]

shivasiddharth, 2018-05-25:
Closing this for inactivity

closed 2018-05-25 by shivasiddharth