shivasiddharth/GassistPi #296: problem with the java client

binarycis, 2018-03-30:
pi@raspberrypi:~/alexa-avs-sample-app/samples/javaclient $ cd /home/pi/alexa-avs-sample-app/samples/javaclient&& sudo mvn exec:exec
Warning: JAVA_HOME environment variable is not set.

[INFO] Scanning for projects...
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building Alexa Voice Service Sample Java Client 20160207.7
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- exec-maven-plugin:1.2.1:exec (default-cli) @ sample-java-client ---
Error: Could not find or load main class com.amazon.alexa.avs.App
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 4.436 s
[INFO] Finished at: 2018-03-30T23:25:24+00:00
[INFO] Final Memory: 8M/19M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.codehaus.mojo:exec-maven-plugin:1.2.1:exec (default-cli) on project sample-java-client: Command execution failed. Process exited with an error: 1 (Exit value: 1) -> [Help 1]

binarycis, 2018-03-30:
Noob so not sure what the proper fix is any guidance greatly appreciated

shivasiddharth, 2018-04-04:
Closing this issue --- Wrong git !

closed 2018-04-04 by shivasiddharth