shivasiddharth/GassistPi #301: Error loading client secret: [Errno 2] No such file or directory

agathaone, 2018-04-02:
please excuse me if this is something obvious but I have been attempting to work through the readme for GassistPi for a pi zero with USB mic and HDMI speaker.
I confirmed all the audio is working ok but then I worked through the steps to install GassistPi until I got to the step where I have to paste back the authorisation code from the browser... then I get this:
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Error: Error loading client secret: [Errno 2] No such file or directory: 'client                                                                             _secret_979520266957-7ltg4ijpf76fjcpkg1hg8n94j022edjd.apps.googleusercontent.com                                                                             .json'.
Run the device tool with --client-secrets or --project-id option.
Or copy the client_secret_979520266957-7ltg4ijpf76fjcpkg1hg8n94j022edjd.apps.goo                                                                             gleusercontent.com.json file in the current directory.

So the google assistant demo mode didn't start and when I try to start it manually using
/home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id 'replace this with your project id'  --device-model-id 'replace this with the model id'

it plays the opening sound, then says hello and then quits

Is it something obvious I got wrong?  Thanks

shivasiddharth, 2018-04-02:
Can you confirm that, you have credentials file in the /home/pi directory ?

agathaone, 2018-04-03:
thank you - yes.
cd /home/pi
ls -al 
shows me credentials.json
Then I can nano credentials.json and it says:
{"installed":{"client_id":"979520266957-7ltg4ijpf76fjcpkg1hg8n94j022edjd.apps.googleusercontent.com","project_id":"oakpi-f4e42","auth_uri":"https://accounts.google.com/o/oauth2/auth","token_uri":"https://accounts.google.com/o/oauth2/token","auth_provider_x509_cert_url":"https://www.googleapis.com/oauth2/v1/certs","client_secret":"C85ruyZWmj2gHro0twTy4tDf","redirect_uris":["urn:ietf:wg:oauth:2.0:oob","http://localhost"]}}

Is it something I got wrong when I registered the project on Actions on Google?

Thanks

Enzoinventor, 2018-04-03:
hi, google has released a new system to create the credentials.json file which is what you copied in home pi.
there are still some updates to be completed in gassistpi but you can still use the project by downloading the credential.json file from here:
https://console.developers.google.com/apis/credentials
select your project and at the bottom right you will find an arrow to start downloading your credential.json which will also contain numbers.
Then paste this new file into home / pi and re-tap the recording by copying the url etc. etc.
Find the info in the readme.

agathaone, 2018-04-03:
I'm so sorry but I didn't much further.  I followed the link and used the blue arrow button to create credentials.   In 'find out what credentials I need' I chose Google Assistant API and it created a service key??

Then I downloaded the 4 different .json files (it wasn't obvious which is the important one or whether they all are).
I copied them over to home/pi and reran the script to install google assistant 
I got this error 
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/flow.py",                                                                              line 147, in from_client_config
    'Client secrets must be for a web or installed app.')
ValueError: Client secrets must be for a web or installed app.

The .json file that I specified for the credentials was 18oakpi-f93a7b3fdf9f.json
and it contained this:
{
  "type": "service_account",
  "project_id": "oakpi-f4e42",
  "private_key_id": "f93a7b3fdf9f480e4da8c8c1d7836605bacf836a",
  "private_key": "-----BEGIN PRIVATE KEY-----\nMIIEvQIBADANBgkqhkiG9w0BAQEFAASCBKcwggSjAgEAAoIBAQCLX8FPkj/CPkHe\n5rBFs+JYbQhImoWdvEjiKT7j64macRojYplL0yhRpjJ$
  "client_email": "oakpi-674@oakpi-f4e42.iam.gserviceaccount.com",
  "client_id": "101169255296458107702",
  "auth_uri": "https://accounts.google.com/o/oauth2/auth",
  "token_uri": "https://accounts.google.com/o/oauth2/token",
  "auth_provider_x509_cert_url": "https://www.googleapis.com/oauth2/v1/certs",
  "client_x509_cert_url": "https://www.googleapis.com/robot/v1/metadata/x509/oakpi-674%40oakpi-f4e42.iam.gserviceaccount.com"

So presumably I chose the wrong json file, but which file should I be looking for?
These files are available to me to download:

Other client 2 | 3 Apr 2018 | Other | 979520266957-qfqhrd45uvgeo0b05jttfqqp3sdoo5a8.apps.googleusercontent.com |  
-- | -- | -- | -- | --
  | oakpi-f4e42-18oakpi-u5twqe | 2 Apr 2018 | Other | 979520266957-7ltg4ijpf76fjcpkg1hg8n94j022edjd.apps.googleusercontent.com |  
  | New Actions on Google App | 2 Apr 2018 | Web application | 979520266957-ii6bvlimftd4j2lism45rd854pe5965q.apps.googleusercontent.com |  
  | Web client (auto created by Google Service) | 2 Apr 2018 | Web application | 979520266957-309thrh2dls3jhikrlivvkq7lt2ae51g.apps.googleusercontent.com

I'm sorry for taking so much of your time.

Enzoinventor, 2018-04-03:
the json file you need to download is only the first one where it is written (other)

Enzoinventor, 2018-04-03:
look at the minute: 5:20
https://www.youtube.com/watch?v=vprJBDDYE8Q&t=484s

agathaone, 2018-04-03:
Thank you.   I followed your video, but where your video started the google assistant demo, my Pi zero gave me:

Enter the authorization code: 4/AAAPIbPXICCRBFVmErBrusDzIjpeBPuVeKG1hkWZLuU79x_3rw_vETg
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Creating new device model
Model oakpi-f4e42-20180403214310 successfully registered
Testing the installed google assistant. Make a note of the generated Device-Id
INFO:root:Connecting to embeddedassistant.googleapis.com
WARNING:root:Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
INFO:root:Registering device
INFO:root:Device registered: 49bfe772-3780-11e8-83bc-b827eb2e57fd
Press Enter to send a new request...
INFO:root:Recording audio request.
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
INFO:root:End of audio request detected
INFO:root:Finished playing assistant response.
Traceback (most recent call last):
  File "/home/pi/env/bin/googlesamples-assistant-pushtotalk", line 11, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 425, in main
    continue_conversation = assistant.assist()
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 231, in wrapped_f
    return self.call(f, *args, **kw)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 313, in call
    start_time=start_time)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 269, in iter
    return fut.result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 398, in result
    return self.__get_result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 357, in __get_result
    raise self._exception
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 316, in call
    result = fn(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 170, in assist
    self.conversation_stream.stop_playback()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 288, in stop_playback
    self._source.stop()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 221, in stop
    self.flush()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 211, in flush
    self._audio_stream.write(b'\x00' * self._flush_size)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1202, in write
    _check(err)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Can't write to an input only stream [PaErrorCode -9974]

But I can nano home/pi/.config/googlesamples-assistant/device_config.json
and it contains:
{"id": "49bfe772-3780-11e8-83bc-b827eb2e57fd", "client_type": "SDK_SERVICE", "model_id": "oakpi-f4e42-20180403214310"}

So I tried /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id oakpi-f4e42 --device-model-id oakpi-f4e42-20180403214310

and it played the google assistant intro sound and said hello and then quit.

Enzoinventor, 2018-04-03:
the video is from: shivasiddharth
have you connected the button to gpio22?

agathaone, 2018-04-03:
Erm.... what button?

Enzoinventor, 2018-04-03:
pushbutton uses sdk Service and to let the assistant speak, you have to press the button that is connected to gpio 22. but have you read the readme file?
To run ok google you have to start the main.py script by assigning the model id as you did for pushbutton

agathaone, 2018-04-03:
I'm so sorry to be a bother to you.
Yes,  I have read the readme file but the distinction between what is possible on Pi zero vs the other models is not clear.   I found only this at the end of readme:

22 | Pushbutton trigger for gRPC API. Connect a pushbutton between GPIO 22 and GRND for manually triggering

Is there another readme/info source that explains about the pushbutton?

shivasiddharth, 2018-04-04:
> So I tried /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id oakpi-f4e42 --device-model-id oakpi-f4e42-20180403214310
The ids should be within quotes. By making a quick google search, you should be able to get the pinouts for Pi Zero. Just pay attention to the fact that, GPIO 22 is different from Pin 22.

shivasiddharth, 2018-05-25:
Closing this for inactivity

closed 2018-05-25 by shivasiddharth