shivasiddharth/GassistPi #303: MagicMirror integretion

serhatkimya, 2018-04-04:
hi everyone. İ try to use  MagicMirror at the same Pi, but i can not working it, at the read me file nothing write explanation , is there any setting which was not written? thanks so much.

shivasiddharth, 2018-04-04:
Ofcourse there is info in the readme about controlling MM using GA. Please refer to [this](https://github.com/shivasiddharth/GassistPi#controlling-magic-mirror-by-voice). For guidance on installing MM, you need to look up the official MM repo.

serhatkimya, 2018-04-04:
i think problem is the ip adress, installed same pi and running MagicMirror but google says mm not online :((

shivasiddharth, 2018-04-25:
if you are using the same pi then try using the ip address itself instead of localhost or vice versa.

shivasiddharth, 2018-05-24:
closing this for inactivity.

closed 2018-05-24 by shivasiddharth