shivasiddharth/GassistPi #309: High CPU usage on pi zero

jaedeezy, 2018-04-12:
Seems to be working as it should but I noticed the CPU usage stays at a constant 98% while memory is at about  16%. Is this normal? I'm running the push-button.service on a pi zero w (raspbian jessie lite). When I stop the service CPU% drops to about 1 or less. I'm new to linux so not sure how to troubleshoot. Any help is greatly appreciated!

jaedeezy, 2018-04-20:
Also, this is only after I changed trigger pin to 23 and.. 

while True:
            if wait_for_user_trigger:
                button_state=GPIO.input(23)

When its set to the default 22 for both, the assistant keeps beeping every few seconds as if I pushed the button and will continue like that indefinitely until I stop the service. With both set to 23, everything works fine aside from CPU% hovering close to 100% and auto rebooting sometimes when I press the button.

shivasiddharth, 2018-04-25:
@jaedeezy  I have clearly asked the users to use the latest stretch version. Can you try the Stretch ?

shivasiddharth, 2018-05-24:
Closing this for inactivity.

closed 2018-05-24 by shivasiddharth