shivasiddharth/GassistPi #316: Getting: google-assistant-library 0.1.1 has requirement google-auth==1.0.1, but you'll have google-auth 1.3.0 which is incompatible.

juggledad, 2018-04-21:
I'm trying to install this with the vVoice Kit v1 and towards the end of running gassist-installer-pi3.sh I see the message
`google-assistant-library 0.1.1 has requirement google-auth==1.0.1, but you'll have google-auth 1.3.0 which is incompatible.`
and after visiting the URL to authorize this application I see
`credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Usage: googlesamples-assistant-devicetool register-model [OPTIONS]

Error: Got unexpected extra arguments (client 1-20180421121435)`

Any thoughts?

shivasiddharth, 2018-04-25:
looks like the model-id that you entered is messed up. The model id is generated using the project id and the time stamp. There seems to be a deviation from this pattern in your case.

shivasiddharth, 2018-05-24:
Closing this for inactivity.

closed 2018-05-24 by shivasiddharth