shivasiddharth/GassistPi #318: Is there an easy way to check GPIO05 and GPIO06 pins status

bionaught, 2018-04-24:
I'd like to connect up an esp and a neopixel, but I'd like to be sure the two pins used are doing what they are supposed to before flashing an 8266 and connecting up the neopixel.. I tried with the Android app "GPIO tool for rasberry pi" and couldn't see any pins doing anything when testing GassistPI.

Great work shivasiddharth... I'll be cringing when I look back at this question after a while, when I'm more familiar with this sort of stuff. (better that, than I chuck my computer across the room :-D)

bionaught, 2018-04-24:
$ watch -n0.125 gpio readall on the command line showed me the answer... both pins are going high and low as expected. Though running the gpio utility crashes GassPi for each question response. She restarts again automatically. 
Now to find my esp8266 after a break  (gonna stop on a success)

shivasiddharth, 2018-05-24:
Closing this for inactivity.

closed 2018-05-24 by shivasiddharth