shivasiddharth/GassistPi #320: Recipes from Edamam not working

Raybenza, 2018-04-25:
Hi
I have entered my id and app key in quotes in the actions.py file. Is that recipeurl correct though? It wont work, Raspberry PI with voice kit keeps rebooting after i give the command " hey google, get ingredients for.....". Everything else is working.

appid='ENTER-YOUR-APPID-HERE'
appkey='ENTER-YOUR-APP-KEY-HERE'
recipeurl = 'https://api.edamam.com/search?q='+item+'&app_id='+appid+'&app_key='+appkey

shivasiddharth, 2018-04-26:
The recipe feature needs the pushbullet feature to be used, that is how the recipes are sent to the phone. Please go through the readme for guidelines on using the pushbullet thingy.

Raybenza, 2018-04-26:
OK I see now, it only pushes the text to your phone... I was expecting it to be read to me by voice on the Raspberry PI Google voice kit.

closed 2018-04-26 by Raybenza