shivasiddharth/GassistPi #323: Triggering from GPIO17 on Zero w

jptrsn, 2018-05-04:
I'm using a Pi Zero W with the ReSpeaker 2mic hat. I've gotten things working, and after running the pi zero installer script, the assistant can hear and respond properly, so I'm confident that the installation was successful.

However, I'm not able to trigger the assistant with the on-board button. I've modified pushbutton.py to change the trigger pin to GPIO17.

```
#Trigger Pin
GPIO.setup(17, GPIO.IN)
```
and
```
 # If no file arguments supplied:
        # keep recording voice requests using the microphone
        # and playing back assistant response using the speaker.
        # When the once flag is set, don't wait for a trigger. Otherwise, wait.
        wait_for_user_trigger = not once
        while True:
            if wait_for_user_trigger:
                button_state=GPIO.input(17)
                if button_state==True:
                    continue
                else:
                    pass
```
However, it's not picking up on button presses. When I reboot the pi, I hear the chime, then the assistant says "Hi" but I get nothing after that. It may be that the logic is inverse from GPIO22 (although I couldn't find any reference to whether the button is active low or active high for either project).

I'm also curious about the support for custom hot words on the pi zero w. I've been reading the issues, and it looks like it was previously supported, but then Google made a change to their API so it's not working at present -- is this a correct statement?

abovedmc, 2018-06-02:
I too am using the ReSpeaker 2michat and I was able to install the assistant and use it during the setup.  Since the reboot I too hear the chime and am greeted by "Hi" in a pleasing female voice.  Beyond that I'm stuck.  I tried to do a manual start and get an error message: "Error: no such option: --XXXXXXX" the X;s are my Project Id.

I too tried using the onboard pushbutton which I understand is on gpio pin 17, but have no clue how to get it connected to the program.  

So close but so far.  I don't understand programming very much so need guidance.

shivasiddharth, 2018-06-04:
I am not sure, if the pushbutton on the HAT is connected to grnd or +3v. Based on that, you may have to change the pull-up or pull-down in the script.

abovedmc, 2018-06-14:
In looking at further, I see GPIO 17  is connected to physical pin 11.  Could you give a little more coaching on which file to change and how it might look?
https://pinout.xyz/pinout/pin11_gpio17

I'm getting close to a solution.  I was able to start pushbutton.py at startup and can manually start it too.  But without a pushbutton, I can't get assistant to respond after the chimes and "hi".

jptrsn, 2018-06-15:
@shivasiddharth Here is a capture of the Respeaker 2 Mic hat schematic, indicating that the button is connected to VCC. Does this mean that we need to invert the logic? The full schematic is [here](http://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/#resources).

[image]

shivasiddharth, 2018-06-25:
Just changing the pin numbers should suffice. As they have added an additional pullup resistor, you may not want to mention that in the script.

Smanar, 2018-06-25:
Try with the command 
`gpio readall`
Don't press button/make command then press button/make command and compare result.

abovedmc, 2018-06-26:
I tried deleting line 97 in pushbutton.py and replacing it with gpio readall, GPIO.readall GPIO (readall) and all returned errors and failed to work.  Anyone have more ideas about using the pushbutton on ReSpeaker 2 mic hat with gassistpi? The only way I can get response is to reinstall everything and on first boot I can communicate with google assistant.

Smanar, 2018-06-27:
Lol
"gpio readall" is a shell command, not python.

abovedmc, 2018-07-06:
Has anyone been able to get the Respeaker 2 mic hat to work with the pushbutton?

shivasiddharth, 2018-09-12:
Closing this for inactivity

closed 2018-09-12 by shivasiddharth