shivasiddharth/GassistPi #326: audio

iamhollis84, 2018-05-05:
chmod +x /home/pi/GassistPi/audio-drivers/USB-DAC-HDMI/scripts/install-usb-mic-hdmi not in directory

jptrsn, 2018-05-05:
You're missing the .sh file extension.

`sudo chmod +x /home/pi/GassistPi/audio-drivers/USB-DAC/scripts/install-usb-dac.sh `

iamhollis84, 2018-05-05:
Will that work if I’m using a usb mic but the hdmi audio?

jptrsn, 2018-05-05:
Ah, was a copy/paste error; should have been:

`sudo chmod +x /home/pi/GassistPi/audio-drivers/USB-MIC-HDMI/scripts/install-usb-mic-hdmi.sh  `

iamhollis84, 2018-05-05:
It says not a directory

jptrsn, 2018-05-05:
Did you install git and clone the repository?

iamhollis84, 2018-05-05:
wget https://github.com/shivasiddharth/GassistPi then apt-get install raspberrypi-kernel. Did I miss something?

shivasiddharth, 2018-05-25:
wget is not the command to clone the repo. Please see the README doc.

closed 2018-05-25 by shivasiddharth