shivasiddharth/GassistPi #328: Can't start google assistant after reboot my pi

aprooooo, 2018-05-05:
Google assistant can't start automatically
And now when I try to start it by ssh
got this error
Connecting to 192.168.1.12 port 22, please wait...
login as: 
pi@192.168.1.12's password: 
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat May  5 18:01:13 2018

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --de
evice_model_id aproo-dcf0e-20180505171947
Failed to import the site module
Traceback (most recent call last):
  File "/usr/lib/python3.5/site.py", line 580, in 
    main()
  File "/usr/lib/python3.5/site.py", line 563, in main
    known_paths = venv(known_paths)
  File "/usr/lib/python3.5/site.py", line 508, in venv
    addsitepackages(known_paths, [sys.prefix])
  File "/usr/lib/python3.5/site.py", line 344, in addsitepackages
    addsitedir(sitedir, known_paths)
  File "/usr/lib/python3.5/site.py", line 212, in addsitedir
    addpackage(sitedir, name, known_paths)
  File "/usr/lib/python3.5/site.py", line 168, in addpackage
    for n, line in enumerate(f):
  File "/usr/lib/python3.5/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe0 in position 1: invalid continuation byte
pi@raspberrypi:~ $

shivasiddharth, 2018-05-05:
The device model id should be within quotes.

aprooooo, 2018-05-06:
Now I have this error 

pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --de
evice_model_id 'aproo-dcf0e-302283925457'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 17, in 
    from kodijson import Kodi, PLAYER_VIDEO
  File "/home/pi/env/lib/python3.5/site-packages/kodijson/__init__.py", line 2, in 
    from .kodijson import *  # NOQA
  File "/home/pi/env/lib/python3.5/site-packages/kodijson/kodijson.py", line 3, in 
    import requests
ValueError: source code string cannot contain null bytes
pi@raspberrypi:~ $

closed 2018-05-06 by aprooooo