shivasiddharth/GassistPi #329: source code string cannot contain null bytes

aprooooo, 2018-05-06:
I have this error 

pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --de
evice_model_id 'aproo-dcf0e-302283925457'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 17, in 
    from kodijson import Kodi, PLAYER_VIDEO
  File "/home/pi/env/lib/python3.5/site-packages/kodijson/__init__.py", line 2, in 
    from .kodijson import *  # NOQA
  File "/home/pi/env/lib/python3.5/site-packages/kodijson/kodijson.py", line 3, in 
    import requests
ValueError: source code string cannot contain null bytes
pi@raspberrypi:~ $

shivasiddharth, 2018-05-06:
Please do not open multiple issues for the same issue... This is the same as #325  The model-id is dubious. I have set the model id to be generated based on time and date and from the looks of it, Your model-id does not seem to be generated by the installer script.

aprooooo, 2018-05-06:
@shivasiddharth
I'm sorry for that 
But The model id generated by script. 
And when I restarted I got a taskbar issues

shivasiddharth, 2018-06-04:
For general issues like those related to taskbar please use the gitter or raspberrypi's forums.

closed 2018-06-04 by shivasiddharth