shivasiddharth/GassistPi #332: Problem with sound card mic

Chetanhiremath, 2018-05-09:
Hi,

I have been working on porting google assistant on RPI3 from past few weeks, thanks for all your scripts and code which made it easy to accomplish task.

After completing all the steps i am getting stuck at this,

ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED 

After these lines i have tried to say "ok google", "Hey google", "put the volume to maximum", everything i tried but it went in vain nothing working(speakers are working properly i have tested them with test script).

Please help me in resolving the above issue.

After sometime i get below error,
[1650:1679:ERROR:audio_input_processor.cc(748)] Input error
[1650:1679:ERROR:audio_input_processor.cc(748)] Input error
[1650:1679:ERROR:audio_input_processor.cc(748)] Input error

Regrads,
Chetan Hiremath

shivasiddharth, 2018-05-09:
This seems like an issue with your mic. May be incompatible mic/non-functional one. This already works on RPi3, so i dont understand as to what you meant by porting..

Chetanhiremath, 2018-05-09:
Yes, it is even working for me , only issue is the one i mentioned above.

I checked with multiple MIC, though the problem remains same.

Chetanhiremath, 2018-05-14:
I am able to fix the problem as the problem was with sound card itself, now i changed the sound card and its working fine.

I am using USB-DAC as i am using sound card but every time i got stuck at below lines 

ON_MUTED_CHANGED: 
{'is_muted': False} 
ON_START_FINISHED

please help me solving this

Chetanhiremath, 2018-05-16:
Can anyone help me out in solving this......

shivasiddharth, 2018-06-04:
Attach your asoundrc and asound.conf files along with the log.

Chetanhiremath, 2018-06-05:
Actually problem got solved 2 days back sorry that i didn't update it here, problem was with my sound card instead i used USB MIC which solved my issue.

Thanks for you help, hereby am closing the issue :).

closed 2018-06-05 by Chetanhiremath