shivasiddharth/GassistPi #333: MagicMirror Module

sup3rgiu, 2018-05-09:
I was wondering if it possible to create a MagirMirror module in order to integrate GassistPi in the MagicMirror interface.

I know that already exixst a module called MMM-GoogleAssistant, but I think that GassistPi offers more features and also run in background with an autostart.

Thanks!

datrix88, 2018-05-10:
I'll be awesome.
UP for this feauture.

bimbim2401, 2018-05-23:
I have install a magic mirror on my PI and install gassitpi also with interaction between us ( hide/show module, send notification with getrequest) It works great but I am agree that a module will be more efficient

shivasiddharth, 2018-05-24:
Unfortunately, i dont have the expertise to create modules. I shall leave this open just in case someone is ready to help.

datrix88, 2018-05-24:
@shivasiddharth I edit the main.py to works with the existent module (MMM-GoogleAssistant) for Magic Mirror.
If you want I can share the code.

bimbim2401, 2018-05-24:
@shivasiddharth  i don't think that create a module will be difficult, i am not an expert and i create 2-3 modules that interpret gassitPI  vocal command. i would love to create a module with gassistPI but i have not your experience of google assistant.

for example i do not understand why i can put french language in MMM- Assistant and why do i have to pay for google cloud speech api and not in your assistant.

If you want i can show you a video of gassitPI in action with magic mirror controling youtube, led,...

@datrix88 i will be great, can you also explains what are the interaction between us

datrix88, 2018-05-25:
MMM-Assistan is not the correct module to use GoogleAssistant with Magic Module. 
Here's the right module:
https://github.com/gauravsacc/MMM-GoogleAssistant

--------------

To adapt GassistPi with MMM-GoogleAssistant is necessary add pubnub api and functions into main.py as into the .py file used and integrate into MMM-GoogleAssistant.

This is a my fork with the integration:

https://github.com/sup3rgiu/GassistPi-IT/blob/master/src/main.py

How you can see there are severale parts of code about pubnub functions.

bimbim2401, 2018-05-30:
can you explain a little bit more what is the goal to have gassisPI and mmm google assistant ?

what are the interaction between us ?  the mmm-google-assistant of the magic mirror or other module can ask gassistPI to say something with a notification ?

shivasiddharth, 2018-06-25:
Closing this for inactivity.

Marcreed48, 2018-10-28:
I believe if you look at all the magic mirror module codes and find out what they have in common, take those common code lines and fill in the blanks with GassistPi.

nabbi95, 2019-02-06:
> @shivasiddharth I edited the main.py to works with the existent module (MMM-GoogleAssistant) for Magic Mirror.
> If you want I can share the code.

please share code and can you please explain it in its easiest way because i tried  many times and couldn't integrate MMM-GoogleAssistant module with GassistPi   
Thank You

icripps, 2019-02-08:
> MMM-Assistant is not the correct module to use GoogleAssistant with Magic Module.
> Here's the right module:
> https://github.com/gauravsacc/MMM-GoogleAssistant
> 
> To adapt GassistPi with MMM-GoogleAssistant is necessary add pubnub api and functions into main.py as into the .py file used and integrate inside MMM-GoogleAssistant.
> 
> This is a my fork with the integration:
> 
> https://github.com/sup3rgiu/GassistPi-IT/blob/master/src/main.py
> 
> How you can see there are several parts of code about pubnub functions.

@datrix88 please can you explain how to integrate the main,py into the assistant.py?

icripps, 2019-02-08:
> please share code and can you please explain it in its easiest way because i tried many times and couldn't integrate MMM-GoogleAssistant module with GassistPi
> Thank You

@nabbi95 i am also stuck here and confused. i don't know what to do. is anyone reading this? please reply

icripps, 2019-02-08:
@shivasiddharth can you help us please? i have seen that you just rolled out a new update for Gassistpi

closed 2019-02-08 by close-issue-app[bot]

shivasiddharth, 2019-02-08:
I have re-opened this issue. Since MM is on a popular demand, will release an update for this.

shivasiddharth, 2019-02-08:
Since its not in issue template, its getting automatically closed..

icripps, 2019-02-08:
> I have re-opened this issue. Since MM is on a popular demand, will release an update for this.

@shivasiddharth okay thank you so much, when should we be expecting this please?

bimbim2401, 2019-02-08:
hello,

I have already made some code for interaction between gassitpi and the magic mirror google assistant module. I will share it this week.

In my code, you can see on the mirror the text your are saying and you can launch gassistpi with the remote magic mirror application.

sup3rgiu, 2019-02-08:
> hello,
> 
> I have already made some code for interaction between gassitpi and the magic mirror google assistant module and i am not using pornhub. I will share it this week.
> 
> In my code, you can see on the mirror the text your are saying and you can launch gassistpi with the remote magic mirror application.
> 
> @shivasiddharth how do I have to give you my code to merge it in the master ?

@bimbim2401 I'm glad you did not use Pornhub :D

shivasiddharth, 2019-02-08:
ROTFL max...

bimbim2401, 2019-02-08:
mdrrrrr sorry pubnub

icripps, 2019-02-09:
> @bimbim2401 I'm glad you did not use Pornhub :D

OMG i laughed at that so hard , lmao

icripps, 2019-02-09:
> mdrrrrr sorry pubnub

@bimbim2401 can you send me your code pleaase , or is there anyway i can contact you

icripps, 2019-02-09:
@shivasiddharth please are you still going to integreate your code into magic mirror?

nabbi95, 2019-02-12:
yes i want to integrate my magicmirror with Gassist pi.
please share if you have some code to do that.

bimbim2401, 2019-02-26:
sorry for the delay, i have got so much work to do, here are the modifications i have made to obtain graphical representation for the gassistpi launch and see the speaking sentence on the magic mirror.

it can be also fix the https://github.com/shivasiddharth/GassistPi/pull/463

First add library :
from urllib.parse import unquote
import socket

then add the websocketslistener class :
class WebSocketsListener(Thread):
    """Thread chargé d'ecouter les messages sur la websocket."""

    def __init__(self, assistant):
        Thread.__init__(self)
        self.websocket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        self.websocket.bind(('127.0.0.1', 8081))
        self.assistant = assistant

    def run(self):
        print ("Open socket")

        while True:
            self.websocket.listen(5)
            client, address = self.websocket.accept()
            print (format( address ))

            response = client.recv(255)
            if response != "":
                print (response)
                str_response = response.decode("utf-8")
                if 'assistant' in str_response:
                    
                    if 'mode' in str_response:
                        if 'start' in str_response:
                            print("start the assistant") 
                            self.assistant.start_conversation()
                        if 'stop' in str_response: 
                            print("stop  the assistant") 
                            self.assistant.stop_conversation()

                    elif 'say' in str_response:
                        token = str_response[str_response.find("say=")+4:str_response.find("HTTP")-1]
                        token = unquote(token)
                        say(token)

        print ("Close socket")
        client.close()
        stock.close()

Then when conversation turn started :
mmreq=requests.get("http://"+mmmip+":8080/assistant?event=ON_CONVERSATION_TURN_STARTED")

when recognizing speech finished :
 mmreq=requests.get("http://"+mmmip+":8080/assistant?event=ON_RECOGNIZING_SPEECH_FINISHED&text="+event.args['text'])

When conversation turn finished :
mmreq=requests.get("http://"+mmmip+":8080/assistant?event=ON_CONVERSATION_TURN_FINISHED")

and last in the def main (), start the listenner :
webSocketsListener = WebSocketsListener(assistant)
webSocketsListener.start()

shivasiddharth, 2019-02-26:
@bimbim2401 What is the module that you are using along with this code ?

bimbim2401, 2019-02-26:
sorry i had some difficulties to attach a file at work, so now i am at home here you can find the module google assistant to put in the magic mirror module folder to work with gassistpi. it will show a image when gassistpi is launch and show the text on the mirror
[MMM-GoogleAssistant.zip](https://github.com/shivasiddharth/GassistPi/files/2907894/MMM-GoogleAssistant.zip)

the thread is use in order to receive order from the remote magic mirror application 
[MMM-Remote-Control.zip](https://github.com/shivasiddharth/GassistPi/files/2907890/MMM-Remote-Control.zip)
to launch the assistant or to make it saying something you want

here is the main.py file with all my modification
[main.txt](https://github.com/shivasiddharth/GassistPi/files/2907884/main.txt)

@shivasiddharth can you please merge my modification in the master ?

icripps, 2019-02-28:
@bimbim2401 please can you make your explanation noob friendly ?