shivasiddharth/GassistPi #334: Problem with nodemcu via webserver

alasalem009, 2018-05-10:
I can't turn on any device through the webserver but when i put the ip adress in the chronuim i can manually turn on/off any device; i put the ip adress like this '192.168.1.8' with no http and used this syntaxe"hey google, trigger turn device1 on" , please i need help it's urgent

iamericha, 2018-07-20:
@alasalem009 

Hi there, I had a problem getting my NodeMCU to respond with the "Trigger turn" keywords.  If you look until issue #280, I provided a tip to get around this.  Here's the text I had provided.  I haven't tried this with the latest update but I think all the functions are still where they should be.
------------------------
I had a hard time getting my NodeMCU to switch with the "Trigger" keyword. I went into your main.py file and added a keyword "node" and had that call your ESP function in the actions.py file directly and it works perfectly now.

For example, my phrase would be: "NODE LAMP ON" and it'll switch the pin I had connected to my lamp on.
------------------------

shivasiddharth, 2018-09-12:
Closing this issue for inactivity.

closed 2018-09-12 by shivasiddharth