shivasiddharth/GassistPi #336: Cast to GassistPI

TypeDaniel, 2018-05-14:
Hi,

I know this was being looked into some time ago, but is it possible to cast to a GassistPi now? It was raised previously 

https://github.com/shivasiddharth/GassistPi/issues/46

Thanks for your time, =)

aandroide, 2018-05-24:
hello recently I wrote to the developers and for now this option is not provided.
In my opinion because they want to launch well in the market google home devices. perhaps in a very distant future it will also come for raspberry.
if in fact you try to assign a room to gassistpi, you will notice that you can not do it and it will tell you error connesione.

shivasiddharth, 2018-05-24:
I can include shairport-sync to be installed with the regular sdk.

TypeDaniel, 2018-05-24:
Hi,

Thanks for getting back to me, that’s a shame! Hopefully in the not so distant future it becomes available.

Appreciate all your hard work you’ve put in