shivasiddharth/GassistPi #341: Google Music Playlist Help

BeatboxHero, 2018-05-20:
Hello and thank you so much for this project! I got it all working last night and I am currently setting up each individual function with my own settings. 

My question is, how do I know which playlist is being selected as the first playlist? I use google music quite often and would like to understand more about calling specific playlists. How do I know what playlist GassistPi finds first? I would be more than willing to work on this and document it for other users, if it isn't already. I just need a little guidance. 

Thanks gain!

shivasiddharth, 2018-05-20:
I have made a detailed video on the Google Play Music feature you can follow that: https://youtu.be/RxGFEPV5wOg if you have any queries even after following that video do let me know.

BeatboxHero, 2018-05-20:
Funny! I actually watched the first part of that video, I just didn't watch it until the end.

Just to verify, from what I understand, I would basically have to add more elif statements to the code? After the first if and before the last else. Something like this?:

```
elif 'second'.lower() in phrase or 'two'.lower() in phrase  or '2'.lower() in phrase:
            os.system('echo "play_playlist(1)\n" >> /home/pi/GassistPi/src/trackchange.py')
            say("Playing songs from your playlist")
            play_playlist(1)
```

That's easy enough, if that is the case. The only problem that I see is that I don't know how the "play_playlist" numbers line up to the playlists in Google music. Can I find that in the playlist.json file? I can find the playlist I am looking for in that file, I just don't know how to tell what number it would be. I am only looking to enable a specific playlist (OR one to two favorite playlists) that I use. So how would I find the specific playlist name and number so that I can write just the code I need?

BeatboxHero, 2018-05-21:
Figured it out the hard way. Just to verify, I did this and it worked like a charm. The only limit here is that you can only have playlists 1-10 (numbers 0-9 in the code) I think. Once I setup 10, 11, and 12 it seemed to loop back to the first playlist again. Thanks again for this project!

shivasiddharth, 2018-05-21:
The playlists are ordered based on the date of creation if i remember right. The issue in your case with 10, 11 and 12 is the presence of 1 in all of them. `'1'.lower() in phrase` code given for playing the first playlist detects the '1' in 10,11,12 and plays the first playlist.

Refer to [this](https://github.com/shivasiddharth/GassistPi/blob/68ddae12c611697e06e29707c8f760941b356212/src/actions.py#L1526) line for extracting the numbers from the user command.

shivasiddharth, 2018-06-14:
Closing this for inactivity.

closed 2018-06-14 by shivasiddharth