shivasiddharth/GassistPi #346: GassistPi Crash on News and low alarm volume.

kodxana, 2018-05-22:
GassistPi restart after asking for news. It crashes evry time i try to ask for news. Crash and then reboot assistant. Also im trying to setum alarm so that it can wake me up but is quietly.

aandroide, 2018-05-24:
even to myself problem as soon as they ask the news goes crashing

shivasiddharth, 2018-05-24:
The issue is, i have assigned news as a custom command. Remove the news part from [here](https://github.com/shivasiddharth/GassistPi/blob/204af99249bcac639951423b21b62a568eb1bce3/src/main.py#L421) or comment out that custom function.

kodxana, 2018-05-24:
Whats about alarm sound.

shivasiddharth, 2018-05-24:
Issue a vocal command to set volume to maximum. Can you check if there is an issue with the alarm volume in the official SDK git? I really dont have control over that.

aandroide, 2018-05-24:
Sid. the user above is correct about the news section.
ok that does not work with the word news but it does not work even with the commands established as top news sports news worls news despite we declared the url in action.py:
worldnews = "http://feeds.bbci.co.uk/news/world/rss.xml"
technews = "http://feeds.bbci.co.uk/news/technology/rss.xml"
topnews = "http://feeds.bbci.co.uk/news/rss.xml"
sportsnews = "http://feeds.feedburner.com/ndtvsports-latest"
quote = "http://feeds.feedburner.com/brainyquote/QUOTEBR"

shivasiddharth, 2018-05-24:
Can you attach the log file ?

aandroide, 2018-05-24:
2018-05-24 22:20:55,204 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-24 22:20:55,212 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-24 22:20:55,449 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None

aandroide, 2018-05-24:
I tried to change the feed url but it crashes anyway. https://www.blitzquotidiano.it/feed/
I specify that the tts pronounce the title before crashing.
It seems a problem of tts

aandroide, 2018-05-24:
other information has been output from the log file:
2018-05-24 22:52:16,082 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-24 22:52:16,094 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-24 22:52:16,442 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-05-24 22:56:30,403 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com
2018-05-24 22:56:30,678 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-05-24 22:56:30,771 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET /translate_a/single?hl=it&q=Blitz+quotidiano&client=t&ie=UTF-8&otf=1&tk=62169.428997&tl=it&sl=auto&dt=at&dt=bd&dt=ex&dt=ld&dt=md&dt=qca&dt=rw&dt=rm&dt=ss&dt=t&oe=UTF-8&ssel=0&tsel=0 HTTP/1.1" 200 None
2018-05-24 22:56:30,775 DEBUG gtts.tts lang: it
2018-05-24 22:56:30,776 DEBUG gtts.tts lang_check: True
2018-05-24 22:56:30,777 DEBUG gtts.tts tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=\\!).|(?<=\\？).|(?<=\\！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z])\\, |\\—|\\\n|\\(|\\:|\\¡|\\[|\\：|\\;|\\‥|\\、|\\，|\\)|\\…|\\。|\\¿|\\،|\\]', re.IGNORECASE) from: [<function tone_marks at 0x7310a588>, <function period_comma at 0x7310a5d0>, <function other_punctuation at 0x7310a618>]>
2018-05-24 22:56:30,777 DEBUG gtts.tts pre_processor_funcs: [<function tone_marks at 0x7310a078>, <function end_of_line at 0x7310a468>, <function abbreviations at 0x7310a4b0>, <function word_sub at 0x7310a4f8>]
2018-05-24 22:56:30,778 DEBUG gtts.tts text: Blitz quotidiano
2018-05-24 22:56:30,778 DEBUG gtts.tts slow: False
2018-05-24 22:56:30,785 DEBUG urllib3.connectionpool Starting new HTTP connection (1): translate.google.com
2018-05-24 22:56:30,860 DEBUG urllib3.connectionpool http://translate.google.com:80 "GET / HTTP/1.1" 302 226
2018-05-24 22:56:30,870 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com
2018-05-24 22:56:31,068 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-05-24 22:56:31,308 DEBUG urllib3.connectionpool Starting new HTTP connection (1): translate.google.com
2018-05-24 22:56:31,380 DEBUG urllib3.connectionpool http://translate.google.com:80 "GET //translate/releases/twsfe_w_20180521_RC03/r/js/desktop_module_main.js HTTP/1.1" 200 165090
2018-05-24 22:56:31,610 DEBUG gtts.lang langs: {'de': 'German', 'et': 'Estonian', 'es-us': 'Spanish (United States)', 'pt': 'Portuguese', 'hi': 'Hindi', 'da': 'Danish', 'pt-pt': 'Portuguese (Portugal)', 'ca': 'Catalan', 'fr': 'French', 'ja': 'Japanese', 'fr-ca': 'French (Canada)', 'lv': 'Latvian', 'en-in': 'English (India)', 'en-nz': 'English (New Zealand)', 'sq': 'Albanian', 'en': 'English', 'bn': 'Bengali', 'el': 'Greek', 'af': 'Afrikaans', 'id': 'Indonesian', 'it': 'Italian', 'no': 'Norwegian', 'en-ng': 'English (Nigeria)', 'en-uk': 'English (UK)', 'eo': 'Esperanto', 'en-tz': 'English (Tanzania)', 'ko': 'Korean', 'fi': 'Finnish', 'nl': 'Dutch', 'hy': 'Armenian', 'km': 'Khmer', 'en-gh': 'English (Ghana)', 'en-za': 'English (South Africa)', 'pt-br': 'Portuguese (Brazil)', 'pl': 'Polish', 'ne': 'Nepali', 'es-es': 'Spanish (Spain)', 'ar': 'Arabic', 'en-ca': 'English (Canada)', 'jw': 'Javanese', 'uk': 'Ukrainian', 'sw': 'Swahili', 'is': 'Icelandic', 'sk': 'Slovak', 'zh-tw': 'Chinese (Mandarin/Taiwan)', 'su': 'Sundanese', 'mk': 'Macedonian', 'en-us': 'English (US)', 'tl': 'Filipino', 'hr': 'Croatian', 'vi': 'Vietnamese', 'ro': 'Romanian', 'hu': 'Hungarian', 'fr-fr': 'French (France)', 'en-au': 'English (Australia)', 'si': 'Sinhala', 'sv': 'Swedish', 'th': 'Thai', 'zh-cn': 'Chinese (Mandarin/China)', 'cy': 'Welsh', 'en-ie': 'English (Ireland)', 'ta': 'Tamil', 'cs': 'Czech', 'bs': 'Bosnian', 'sr': 'Serbian', 'es': 'Spanish', 'en-gb': 'English (UK)', 'tr': 'Turkish', 'ru': 'Russian', 'la': 'Latin', 'en-ph': 'English (Philippines)'}
2018-05-24 22:56:31,612 DEBUG gtts.tts pre-processing: <function tone_marks at 0x7310a078>
2018-05-24 22:56:31,615 DEBUG gtts.tts pre-processing: <function end_of_line at 0x7310a468>
2018-05-24 22:56:31,616 DEBUG gtts.tts pre-processing: <function abbreviations at 0x7310a4b0>
2018-05-24 22:56:31,621 DEBUG gtts.tts pre-processing: <function word_sub at 0x7310a4f8>
2018-05-24 22:56:31,622 DEBUG gtts.tts text_parts: 1
2018-05-24 22:56:31,630 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com
2018-05-24 22:56:31,818 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-05-24 22:56:31,836 DEBUG gtts.tts payload-0: {'q': 'Blitz quotidiano', 'client': 'tw-ob', 'ttsspeed': 1, 'ie': 'UTF-8', 'tl': 'it', 'idx': 0, 'tk': '62169.428997', 'textlen': 16, 'total': 1}
2018-05-24 22:56:31,845 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com
2018-05-24 22:56:32,050 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET /translate_tts?q=Blitz+quotidiano&client=tw-ob&ttsspeed=1&ie=UTF-8&tl=it&idx=0&tk=62169.428997&textlen=16&total=1 HTTP/1.1" 200 5280
2018-05-24 22:56:32,058 DEBUG gtts.tts headers-0: {'Referer': 'http://translate.google.com/', 'Connection': 'keep-alive', 'Accept-Encoding': 'gzip, deflate', 'Accept': '*/*', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36'}
[... cut ...]
  File "/home/pi/GassistPi/src/actions.py", line 348, in feed
    while GPIO.input(23):
RuntimeError: You must setup() the GPIO channel first
2018-05-24 22:56:43,309 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-24 22:56:43,315 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-24 22:56:43,576 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None

aandroide, 2018-05-24:
@kodxana 
here is the solution to the problem:
what pin do you use for the stop button? 
in my case I use the 26 and I forgot to edit in actions.py this line: while GPIO.input (23):
with while GPIO.input (26):
and also the line if not GPIO.input (23): with if not GPIO.input (26):
now I do not know in your case what pin you're using for the stop, but I've solved the problem this way.

aandroide, 2018-05-24:
for what concerns the volume of the alarm clock we can not do anything because the raspberry can not be added to the home app and the volume adjustment alarms is only possible from google home devices read here:[https://support.google.com/googlehome/answer/7550530?hl=it&ref_topic=7549809](url)

shivasiddharth, 2018-05-24:
@aandroide , So the gtts is unable to convert the Italian symbols and hence crashing?

aandroide, 2018-05-24:
@shivasiddharth 
No sid. the problem was that I had not changed the gpio stop of the news. I use pin 26 for the stop and in the script it was 23

shivasiddharth, 2018-05-24:
I am considering the issue closed then.

closed 2018-05-24 by shivasiddharth