shivasiddharth/GassistPi #347: YouTube streaming suddenly stopped

VH-Tech, 2018-05-22:
Hi ,
I have been trying to stream music from YouTube from past couple of days and it doesn't do anything after saying "getting YouTube link".
Earlier it used to work amazingly but I don't know what happened

shivasiddharth, 2018-05-24:
Only if you can provide the log file, i would be in a position to help.

aandroide, 2018-05-31:
in the temp folder there is the file gassistpi.log

amitsrivastava79, 2018-06-06:
Hi,

Yesterday Night instead of my manual setup of Google Home on Pi, I used your automated scripts Gassist especially for Youtube stream, but I am also facing the same issue.

I was able to play Dil Se Rai song somehow but its audio quality, as well as volume, was so bad that it's hardly audible.

I will share the logs today evening if you can help ... Thanks in Advance.

amitsrivastava79, 2018-06-06:
@aandroide  Please find the attached log file for reference, while retesting I found a great thing even with the stream command it's actually playing the video on my chromecast and some time there is audio on the SPkrs connected to PI 3,5 jack .. so it seems some setting or configuration issue as the stream command should not cast the video on Chromecast
[Error.txt](https://github.com/shivasiddharth/GassistPi/files/2077092/Error.txt)

shivasiddharth, 2018-06-06:
@amitsrivastava79 This program does nothing to your audio and Pi's 3.5mm jack is controlled by PWM and if you have a bad power supply, audio can be bad. So lets not talk about audio quality.
Unfortunately, your log file is truncated. Definitely the program would have progressed beyond the "Getting Youtube links" audio prompt, but your log file is truncated conveying very little info.

VH-Tech, 2018-06-07:
hi ,
here's my log file
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/2079052/GassistPi.log)

shivasiddharth, 2018-06-07:
If you use up your daily API quota for YouTube, you will hit an error. Other than that, i dont see anything wrong in the Logs.

shivasiddharth, 2018-06-08:
@VH-Tech , your log file is truncated. i can say things from only what i can see. I have made a number of changes to the project over a last couple of days. I can only suggest you to pull the git again and try.

VH-Tech, 2018-06-08:
ok, I will give it a try.

VH-Tech, 2018-06-12:
Hi,
another git pull didn't do it for me.
Instead, I also found that even after deleting your application google assistant was still running.

But I needed to go for a completely new install of raspbian and gassistpi to get it working.

thanks for your precious help buddy

closed 2018-06-12 by VH-Tech