shivasiddharth/GassistPi #352: Magic Mirror Remote Control not working

BluishMechanismz, 2018-05-24:
Hi! I was trying to use the MM portion of the tool, and it does nothing when I put my Pi's IP in. Is there anything I have to do other than putting the IP in? And what would the format of the IP be in? For example, with the Port or just the actual IP itself?

Thank you!

bimbim2401, 2018-05-24:
hi 

No port Just put only the IP and You need to implement the command in the remotecontrol.js file

I can send you my example when I come back home

BluishMechanismz, 2018-05-24:
@bimbim2401 That would be perfect! Thank you!

bimbim2401, 2018-05-24:
hi,
here is the format for the Ip
mmmip='192.XXX.XX.XX'

and i just lokk you don't have anything to do more it normaly work directly with this in the nodehelper.js of your remote control folder

this.expressApp.get("/remote", function (req, res) {
			var query = url.parse(req.url, true).query;

			if (query.action)
			{
				var result = self.executeQuery(query, res);
				if (result === true) {
					return;
				}
			}
			res.send({"status": "error", "reason": "unknown_command", "info": "original input: " + JSON.stringify(query)});
		});

BluishMechanismz, 2018-05-24:
So from what I can get, it works flawlessly with turning the monitor off and on, but I can't disable or enable modules over voice. Anything I have to do to get that working?

shivasiddharth, 2018-05-24:
As of today, i have not added custom actions to enable or disable modules. If some one can provide me a syntax or example, i shall add that functionality.

bimbim2401, 2018-05-24:
You cannot make it identically for all user because it depend of the place of the module you want to hide/show in your config file

As you can see in the example of gassitpi (my config) 
http://"+mmmip+":8080/remote?action=SHOW&module=module_2_currentweather"

Current weather is the second module in the list

If i am right You can get all the module identifier with this in a web browser

http://"+mmmip+":8080/remote?action=MODULE_DATA

Let me know if it s Work