shivasiddharth/GassistPi #356: pushbutton.py always restarting

bimbim2401, 2018-05-25:
Hi,

i have try to make the pushbutton.py working but whitout succes, it always restarting, can you please help me ?

here is the status of the service : 

sudo systemctl status gassistpi-push-button.service
● gassistpi-push-button.service - Google Assistant-Pushbutton
   Loaded: loaded (/lib/systemd/system/gassistpi-push-button.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2018-05-25 20:29:34 CEST; 8s ago
  Process: 6229 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id project-id --device-model-id saved-model-id (code=killed, signal=TE
 Main PID: 6229 (code=killed, signal=TERM)

mai 25 20:29:34 magicPI systemd[1]: gassistpi-push-button.service: Service hold-off time over, scheduling restart.
mai 25 20:29:34 magicPI systemd[1]: Stopped Google Assistant-Pushbutton.
mai 25 20:29:34 magicPI systemd[1]: Started Google Assistant-Pushbutton.
mai 25 20:29:34 magicPI systemd[1]: Stopping Google Assistant-Pushbutton...
mai 25 20:29:34 magicPI systemd[1]: Stopped Google Assistant-Pushbutton.
...skipping...
● gassistpi-push-button.service - Google Assistant-Pushbutton

# And here are the gassistPI log

2018-05-25 20:29:13,438 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-25 20:29:13,445 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-25 20:29:13,693 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-05-25 20:29:13,721 INFO root Connecting to embeddedassistant.googleapis.com
2018-05-25 20:29:14,776 ERROR __main__ Error opening RawStream: Device unavailable [PaErrorCode -9985]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 778, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 669, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 187, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1256, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 772, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]
2018-05-25 20:29:22,634 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-25 20:29:22,643 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-25 20:29:22,894 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-05-25 20:29:22,945 INFO root Connecting to embeddedassistant.googleapis.com
2018-05-25 20:29:24,011 ERROR __main__ Error opening RawStream: Device unavailable [PaErrorCode -9985]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 778, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 669, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 187, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1256, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 772, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]
2018-05-25 20:29:31,816 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-25 20:29:31,824 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-25 20:29:32,064 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-05-25 20:29:32,093 INFO root Connecting to embeddedassistant.googleapis.com
2018-05-25 20:29:33,175 ERROR __main__ Error opening RawStream: Device unavailable [PaErrorCode -9985]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 778, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 669, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 187, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1256, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 772, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]
2018-05-25 20:31:18,049 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-25 20:31:18,057 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-25 20:31:18,300 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None

aandroide, 2018-05-25:
hi there were some errors in the script. you can try to use this code:
https://github.com/aandroide/GassistPi/blob/a06befae4e0edfcbb767ca9c5ada562e238e26dc/src/pushbutton.py
remember to restart gassistpi when you copy and paste and save the new code.

shivasiddharth, 2018-05-25:
The file has been updated. Can you please try replacing the existing file with the file from git.

bimbim2401, 2018-05-25:
I have already before writing my issue but I have some errors with the import browser for the browser_helpers

aandroide, 2018-05-25:
the file was corrected 1 hour after you posted your problem so if you look in the git the file has been updated 30 minutes ago. you need to update the pushbutton.py file again.

bimbim2401, 2018-05-25:
No just only copy paste the pushbutton file. Do I have to update the entire project ? If so is there a minimalist method to do that ?

aandroide, 2018-05-25:
I advise you to start from the beginning, there have been several changes at the code level and to avoid incurring other problems it is better to start from scratch.

bimbim2401, 2018-05-26:
just finish to update all project ans same problem

here is the log 

2018-05-26 12:04:59,739 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-26 12:04:59,748 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-26 12:04:59,999 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-05-26 12:05:00,049 INFO root Connecting to embeddedassistant.googleapis.com
2018-05-26 12:05:01,132 ERROR __main__ Error opening RawStream: Device unavailable [PaErrorCode -9985]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 804, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 681, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 190, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1256, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 772, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]
2018-05-26 12:05:09,089 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-26 12:05:09,099 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-26 12:05:09,341 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-05-26 12:05:09,369 INFO root Connecting to embeddedassistant.googleapis.com
2018-05-26 12:05:10,451 ERROR __main__ Error opening RawStream: Device unavailable [PaErrorCode -9985]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 804, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 681, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 190, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1256, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 772, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]
2018-05-26 12:05:18,500 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-26 12:05:18,513 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-26 12:05:18,772 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-05-26 12:05:18,800 INFO root Connecting to embeddedassistant.googleapis.com
2018-05-26 12:05:19,891 ERROR __main__ Error opening RawStream: Device unavailable [PaErrorCode -9985]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 804, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 681, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 190, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1256, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 772, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]

what is device unavailable ? how can i fix it ? it  works greats with the "
ok google"

aandroide, 2018-05-26:
they are the same mistakes as last night.
you have not updated the project correctly.
You must delete the gassistpi folder from the home / pi path.
and re-download the project

aandroide, 2018-05-27:
what raspberry do you use?
The correct command to lancaiare is not: / home / pi / env / bin / python -u /home/pi/GassistPi/src/pushbutton.py
you must use: / home / pi / env / bin / python -u /home/pi/GassistPi/src/pushbutton.py --project-id 'replace this with your project id' --device-model-id 'replace this with the model id '
Enter the ID of your project and the ID of the model in quotes in the places mentioned.
I right now I installed on a new sd card creating a new project and everything works properly.
Try to create a new project from scratch and get new credentials by registering a new device and repeat the procedure.

bimbim2401, 2018-05-27:
i use a Pi 3, 

i have try with the command you write but it stop immediately with the same error in the log "device unavailable in port audio"
i am already with a new project and with new credentia but i can test to create another on e from the beginning, i have no more idea for the moment.

Do you know what does this erroe "device unavailable" means ? 

Thanks for your help. i really want to fix this.

aandroide, 2018-05-27:
can you try the pushbutton file of my branch? find it here:
https://github.com/aandroide/GassistPi/blob/7863a136f2972bb6d969d6a0c005d3672625a839/src/pushbutton.py

aandroide, 2018-05-27:
https://thepihut.com/products/justboom-amp-hat
Features
Full high quality audio - up to 192kHz / 32bit
2 x 55 Watt peak output at 8 ohms (2 x 30 Watt RMS)
Includes both a DAC and power amplifier - just connect to your passive speakers
Back-powers the Raspberry Pi over the GPIO (with full HAT compliant protection) at 2.5A so only one power supply required for the whole system.
Plug and play compatibility for ease of use
Hardware and software volume control from your Raspberry Pi 
Onboard, hardware jumpers for configuring mute and gain settings (jumpers could optionally be changed with switches)
Mute/enable with GPIO22 (this can be overridden by using jumper J4)

this is a problem because the gassistpi trigger is on pin 22 you have to change pin

bimbim2401, 2018-05-27:
yes I know i have already change it to pin 6 ;-) 

i try you file, just let me two second

shivasiddharth, 2018-05-27:
@bimbim2401 , @aandroide  just noticed something. The Project-id and model-id have not been replaced in the service file

> ● gassistpi-push-button.service - Google Assistant-Pushbutton
Loaded: loaded (/lib/systemd/system/gassistpi-push-button.service; enabled; vendor preset: enabled)
Active: inactive (dead) since Fri 2018-05-25 20:29:34 CEST; 8s ago
Process: 6229 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id project-id --device-model-id saved-model-id (code=killed, signal=TE
Main PID: 6229 (code=killed, signal=TERM)

bimbim2401, 2018-05-30:
@shivasiddharth can you please take a while to look to this issue, i do not think that it comes from a bad installation, you said in issue https://github.com/shivasiddharth/GassistPi/issues/188 it comes from the sdk, and after some spend time on internet, some people said it can comes from the mic and other from the asound format or the stream which need to be stop 

" close stream on stop_recording
  restart stream on start_playback
  flush stream on stop_playback

The previous behaviour (keeping the stream open the whole time)
exposed a race condition where we tried to flush (write to) a stream
that was still recording (read-only mode)."

take a look in https://github.com/googlesamples/assistant-sdk-python/pull/188

https://github.com/googlesamples/assistant-sdk-python/issues/155
https://github.com/googlesamples/assistant-sdk-python/issues/219

bimbim2401, 2018-05-30:
### sudo nano .asoundrc
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

### sudo nano /etc/asound.conf
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

 ### aplay -l
**** Liste des Périphériques Matériels PLAYBACK ****
carte 0: sndrpijustboomd [snd_rpi_justboom_dac], périphérique 0: JustBoom DAC HiFi pcm512x-hifi-0 []
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0

### arecord -l
**** Liste des Périphériques Matériels CAPTURE ****
carte 1: CameraB409241 [USB Camera-B4.09.24.1], périphérique 0: USB Audio [USB Audio]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0

shivasiddharth, 2018-06-03:
I have fixed all issues with pushbutton with the latest updates and have tested everything and found it to be working. The only issue i can think of is your physical button connection. It should be connected between GPIO22 and Ground. Not between GPIO22 and +3v. 
Also please pay attention to the pin number. GPIO22 is not same as Pin22. 
Also check, if your Justboom HAT uses GPIO22 for some other purpose.

bimbim2401, 2018-06-04:
I have already change the GPIO in use for triggering with GPIO6, i use pigpiod in my project, maybe there is an interaction between the state of the pin but i have the same error when i kill pigpiod.

don't worry for this issue, as i succeed to connect the magic mirror with gassistPI in order to launch it with a websockect connection ( see issue https://github.com/shivasiddharth/GassistPi/issues/355 )

shivasiddharth, 2018-06-04:
Closing this then..

closed 2018-06-04 by shivasiddharth