shivasiddharth/GassistPi #361: arecord:main:722: audio open error: File description in bad state

syedazharali880, 2018-05-28:
Bug description :
i m using google assistant on my smart mirror.when i boot OS  assistant work fine but after couple of sec screen blink and assistant cant hear my voice...when i check microphone it show error "arecord: main:722: audio open error: File description in bad state"
when i manually start assistant it show "unable to open slave" "input device could not be opened: default Aborted"

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
No

 2. Is the issue related to assistant installation ?
Yes

 3. Does the assistant service start normally?
Yes

 4. Is the assistant service restarting automatically?
Yes

 Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
when i manually start assistant it show "unable to open slave" "input device could not be opened: default Aborted"
```

### Paste the contents of the terminal below
```
ALSA lib pcm_dsnoop.c:666:(send_pcm_dsnoop_open) unable to open slave
[FATAL:audio_input_streem.cc(47)] Input device could not be opened: default Aborted
```

syedazharali880, 2018-05-28:
i guess i have a issue in .asoundrc
configurations are:

pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
             pcm "hw:1,0"
            channels 1
             }    
}
pcm.!default {
          type asym
          playback.pcm {
                   type plug
                    slave.pcm "hw:0,0"
                      }
          capture.pcm {
                  type.plug
                  slave.pcm "dsnooper"
                      }
}

shivasiddharth, 2018-05-28:
What is your audio config setup ?  What is your MIC and Audio out device?

syedazharali880, 2018-05-28:
i m using USB mic and USB speakers

shivasiddharth, 2018-05-28:
If you are using a USB Mic and USB speaker, then the configuration you used is not right. Why would you want to use the onboard jack configuration, if you are using a USB speaker ? Also, if ou are using multiple USB audio devices, the device card ids will change. Its not like always the USB Mic is 1 and USB Speaker is 2.

Open a terminal and run:
aplay -l
Note the card ids for speaker.
arecord -l
Note the card ids for mic.
Now change the ids in .asoundrc and asound.conf

syedazharali880, 2018-05-29:
now every thing is fine bt this time assistant stop working after a blink. when i boot up ok google work fine bt after 2,3 questions screen blinks n assistant stop working.
i tried to manually start by executing this : 
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'replace this with the model id'
and error is:
ALSA lib pcm_dsnoop.c:666:(send_pcm_dsnoop_open) unable to open slave
[FATAL:audio_input_streem.cc(47)] Input device could not be opened: default Aborted

syedazharali880, 2018-05-30:
zebronics prime 2 2 0 multimedia speakers
3.5mm jack and powered by usb.
and usb microphone

shivasiddharth, 2018-05-30:
I had asked you to post your aplay -l and arecord -l.

syedazharali880, 2018-05-31:
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

syedazharali880, 2018-06-02:
@shivasiddharth

shivasiddharth, 2018-06-04:
This could be an issue with your mic. Try changing the mic.

shivasiddharth, 2018-06-24:
Closing this for inactivity.

closed 2018-06-24 by shivasiddharth