shivasiddharth/GassistPi #363: Google Music doesn't work

leavmealonepy, 2018-05-28:
### Describe the bug:
```
Hey Google, Refresh Music is working but when I want to play music using:
-----Hey Google, play Arctic Monkeys
or
-----Hey Google, play Brianstorm

i got message, ,,I am affraid that something went wrong, when you are ready, get it another try."

with command ,,Hey Google, play Arctic Monkeys on Google Music" I have no response.

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RENDER_RESPONSE:
{
  "text": "I'm afraid something went wrong. When you're ready, give it another try.",
  "type": 0
}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Arctic Monkeys"}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play brianstorm"}
ON_RENDER_RESPONSE:
{
  "text": "Sorry, something's gone wrong. When you're ready, give it another try.",
  "type": 0
}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Arctic Monkeys on Google Music"}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

```
2018-05-28 20:44:26,396 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-28 20:44:26,409 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-28 20:44:26,657 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-05-28 20:45:25,753 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-05-28 20:45:25,766 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-05-28 20:45:26,031 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
```

shivasiddharth, 2018-05-28:
This is a network related issue. Slow internet/Firewall/No internet. As the assistant says, try it again after sometime. This error is a common occurrence when some device in your network is doing heavy downloading or streaming.

leavmealonepy, 2018-05-28:
Connected this time by Ethernet cable, my network is 100mb/s DOWN and 40mb/s UP, still doesn't work.

shivasiddharth, 2018-05-28:
1. You have got the syntax wrong.
2. I am not sure if Arctic Monkeys is an album or a track. If it is a track, then i am afraid it will not work. I have not added the functions to pick a song.
For more details on what you can do, please refer to [this](https://github.com/shivasiddharth/GassistPi#what-you-can-do) part in the readme.

Also, for the streaming to work, you should have your music uploaded to the google music cloud.

leavmealonepy, 2018-05-28:
Arctic Monkeys is Author, I have tried by track too (Brianstorm is track title),
Hey Google, Play all the songs from Google Music crashed my asistant but its closer to work.

shivasiddharth, 2018-05-28:
`Hey Google, Play all the songs from Google Music` is the correct syntax. if it crashed your assistant, then it means that the credentials/login for Google Music have not been provided correctly. Please go through the README for guidelines to setup this feature.

leavmealonepy, 2018-05-28:
```
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play songs by artist Arctic Monkeys from Google Music"}
arctic monkeys
Looking for songs rendered by the artist
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of say.mp3 finished.
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
100

```
And log:
```
2018-05-29 00:05:48,220 DEBUG gmusicapi.Mobileclient1 GetStreamUrl(args=['6aa4e56a-8b83-38b0-b7c8-7ee6fc724058', 'B827EB08F619', 'hi'], kwargs={})
2018-05-29 00:05:48,353 DEBUG urllib3.connectionpool https://mclients.googleapis.com:443 "GET /music/mplay?net=mob&dv=0&opt=hi&hl=en_US&pt=e&slt=1527545148222&sig=7trepxpXKhXUW8Ke_96VvK2gAlM&songid=6aa4e56a-8b83-38b0-b7c8-7ee6fc724058&tier=aa HTTP/1.1" 403 None
2018-05-29 00:05:48,389 ERROR __main__ GetStreamUrl: 403 Client Error: Forbidden for url: https://mclients.googleapis.com/music/mplay?net=mob&dv=0&opt=hi&hl=en_US&pt=e&slt=1527545148222&sig=7trepxpXKhXUW8Ke_96VvK2gAlM&songid=6aa4e56a-8b83-38b0-b7c8-7ee6fc724058&tier=aa
(requests kwargs: {'url': 'https://mclients.googleapis.com/music/mplay', 'headers': {'X-Device-ID': 'B827EB08F619', 'Authorization': '<omitted>'}, 'method': 'GET', 'params': {'net': 'mob', 'dv': 0, 'opt': 'hi', 'hl': 'en_US', 'pt': 'e', 'slt': '1527545148222', 'sig': b'7trepxpXKhXUW8Ke_96VvK2gAlM', 'songid': '6aa4e56a-8b83-38b0-b7c8-7ee6fc724058', 'tier': 'aa'}, 'allow_redirects': False})
(response was: '<HTML>\n<HEAD>\n<TITLE>Forbidden</TITLE>\n</HEAD>\n<BODY BGCOLOR="#FFFFFF" TEXT="#000000">\n<H1>Forbidden</H1>\n<H2>Error 403</H2>\n</BODY>\n</HTML>\n')
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/protocol/shared.py", line 218, in perform
    response.raise_for_status()
  File "/home/pi/env/lib/python3.5/site-packages/requests/models.py", line 935, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://mclients.googleapis.com/music/mplay?net=mob&dv=0&opt=hi&hl=en_US&pt=e&slt=1527545148222&sig=7trepxpXKhXUW8Ke_96VvK2gAlM&songid=6aa4e56a-8b83-38b0-b7c8-7ee6fc724058&tier=aa

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 525, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 519, in main
    gmusicselect(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1165, in gmusicselect
    play_artist(artist)
  File "/home/pi/GassistPi/src/actions.py", line 1084, in play_artist
    streamurl=api.get_stream_url(tracks[currenttrackid])
  File "<decorator-gen-102>", line 2, in get_stream_url
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/utils/utils.py", line 293, in wrapper
    return function(*args, **kw)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/clients/mobileclient.py", line 393, in get_stream_url
    return self._make_call(mobileclient.GetStreamUrl, song_id, device_id, quality)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/clients/shared.py", line 84, in _make_call
    return protocol.perform(self.session, self.validate, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/protocol/shared.py", line 226, in perform
    raise CallFailure(err_msg, call_name)
gmusicapi.exceptions.CallFailure: GetStreamUrl: 403 Client Error: Forbidden for url: https://mclients.googleapis.com/music/mplay?net=mob&dv=0&opt=hi&hl=en_US&pt=e&slt=1527545148222&sig=7trepxpXKhXUW8Ke_96VvK2gAlM&songid=6aa4e56a-8b83-38b0-b7c8-7ee6fc724058&tier=aa
(requests kwargs: {'url': 'https://mclients.googleapis.com/music/mplay', 'headers': {'X-Device-ID': 'B827EB08F619', 'Authorization': '<omitted>'}, 'method': 'GET', 'params': {'net': 'mob', 'dv': 0, 'opt': 'hi', 'hl': 'en_US', 'pt': 'e', 'slt': '1527545148222', 'sig': b'7trepxpXKhXUW8Ke_96VvK2gAlM', 'songid': '6aa4e56a-8b83-38b0-b7c8-7ee6fc724058', 'tier': 'aa'}, 'allow_redirects': False})
(response was: '<HTML>\n<HEAD>\n<TITLE>Forbidden</TITLE>\n</HEAD>\n<BODY BGCOLOR="#FFFFFF" TEXT="#000000">\n<H1>Forbidden</H1>\n<H2>Error 403</H2>\n</BODY>\n</HTML>\n')
```
If credentials/login for Google Music is not correct, how I have synchronised my music by ,,Synchronize Music" command? I have checked it and songs file in pi folder contains correct albums.

leavmealonepy, 2018-05-28:
Is there possibility to get commands shorter and more inteligent?
Like ,,Hey Google, play Arctic Monkeys Brianstorm" or ,,Hey Google, play album AM"
not
,,Hey Google, play songs by artist Arctic Monkeys" and its even not possible to select correct track.

leavmealonepy, 2018-05-28:
Errors from ,,Play all the songs from Google Music" command
```
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play all the songs from Google Music"}
Playing all your songs
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
100

```
and log:

```
2018-05-29 00:14:47,159 DEBUG gmusicapi.Mobileclient1 GetStreamUrl(args=['55aa0cb6-bc11-3294-9874-3e59f30c2e16', 'B827EB08F619', 'hi'], kwargs={})
2018-05-29 00:14:47,284 DEBUG urllib3.connectionpool https://mclients.googleapis.com:443 "GET /music/mplay?slt=1527545687160&sig=5xB92c6Nfn4X6xqrKbDdIJ8dPi4&dv=0&net=mob&songid=55aa0cb6-bc11-3294-9874-3e59f30c2e16&pt=e&tier=aa&opt=hi&hl=en_US HTTP/1.1" 403 None
2018-05-29 00:14:47,523 ERROR __main__ GetStreamUrl: 403 Client Error: Forbidden for url: https://mclients.googleapis.com/music/mplay?slt=1527545687160&sig=5xB92c6Nfn4X6xqrKbDdIJ8dPi4&dv=0&net=mob&songid=55aa0cb6-bc11-3294-9874-3e59f30c2e16&pt=e&tier=aa&opt=hi&hl=en_US
(requests kwargs: {'allow_redirects': False, 'headers': {'Authorization': '<omitted>', 'X-Device-ID': 'B827EB08F619'}, 'url': 'https://mclients.googleapis.com/music/mplay', 'params': {'slt': '1527545687160', 'sig': b'5xB92c6Nfn4X6xqrKbDdIJ8dPi4', 'dv': 0, 'net': 'mob', 'songid': '55aa0cb6-bc11-3294-9874-3e59f30c2e16', 'pt': 'e', 'tier': 'aa', 'opt': 'hi', 'hl': 'en_US'}, 'method': 'GET'})
(response was: '<HTML>\n<HEAD>\n<TITLE>Forbidden</TITLE>\n</HEAD>\n<BODY BGCOLOR="#FFFFFF" TEXT="#000000">\n<H1>Forbidden</H1>\n<H2>Error 403</H2>\n</BODY>\n</HTML>\n')
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/protocol/shared.py", line 218, in perform
    response.raise_for_status()
  File "/home/pi/env/lib/python3.5/site-packages/requests/models.py", line 935, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://mclients.googleapis.com/music/mplay?slt=1527545687160&sig=5xB92c6Nfn4X6xqrKbDdIJ8dPi4&dv=0&net=mob&songid=55aa0cb6-bc11-3294-9874-3e59f30c2e16&pt=e&tier=aa&opt=hi&hl=en_US

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 525, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 519, in main
    gmusicselect(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1109, in gmusicselect
    play_songs()
  File "/home/pi/GassistPi/src/actions.py", line 995, in play_songs
    streamurl=api.get_stream_url(tracks[currenttrackid])
  File "<decorator-gen-102>", line 2, in get_stream_url
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/utils/utils.py", line 293, in wrapper
    return function(*args, **kw)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/clients/mobileclient.py", line 393, in get_stream_url
    return self._make_call(mobileclient.GetStreamUrl, song_id, device_id, quality)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/clients/shared.py", line 84, in _make_call
    return protocol.perform(self.session, self.validate, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/protocol/shared.py", line 226, in perform
    raise CallFailure(err_msg, call_name)
gmusicapi.exceptions.CallFailure: GetStreamUrl: 403 Client Error: Forbidden for url: https://mclients.googleapis.com/music/mplay?slt=1527545687160&sig=5xB92c6Nfn4X6xqrKbDdIJ8dPi4&dv=0&net=mob&songid=55aa0cb6-bc11-3294-9874-3e59f30c2e16&pt=e&tier=aa&opt=hi&hl=en_US
(requests kwargs: {'allow_redirects': False, 'headers': {'Authorization': '<omitted>', 'X-Device-ID': 'B827EB08F619'}, 'url': 'https://mclients.googleapis.com/music/mplay', 'params': {'slt': '1527545687160', 'sig': b'5xB92c6Nfn4X6xqrKbDdIJ8dPi4', 'dv': 0, 'net': 'mob', 'songid': '55aa0cb6-bc11-3294-9874-3e59f30c2e16', 'pt': 'e', 'tier': 'aa', 'opt': 'hi', 'hl': 'en_US'}, 'method': 'GET'})
(response was: '<HTML>\n<HEAD>\n<TITLE>Forbidden</TITLE>\n</HEAD>\n<BODY BGCOLOR="#FFFFFF" TEXT="#000000">\n<H1>Forbidden</H1>\n<H2>Error 403</H2>\n</BODY>\n</HTML>\n')
```

leavmealonepy, 2018-05-28:
So should I do something like:
logged_in = api.login('gmail', 'my app password', Mobileclient.MyPhoneDeviceID)
?

shivasiddharth, 2018-05-28:
I dont want to make random guesses. That will take us nowhere and will be a mere waste of time. I will try and find a fix over this weekend and then update the project.

leavmealonepy, 2018-05-28:
It looks like I have it fixed.

`logged_in = api.login('Vailid GMAIL', 'PASSWORD/APP PASSWORD', 'random numbers')`

then manualy start GassistPi by using command from /home/pi/GassistPi/systemdgassistpi-ok-google.service with exec=

Then in console window you should get information that your deviceID is not correct, and GmusicApi should provide you the correct one, copy it and paste in actions.py

`logged_in = api.login('Vailid GMAIL', 'PASSWORD/APP PASSWORD', 'Correct Device ID from console window')`

leavmealonepy, 2018-05-28:
It would be nice to make gmusic commands more easy and more inteligent in the future.

leavmealonepy, 2018-05-28:
And it looks like gapi supports track names:

```
library = api.get_all_songs()
sweet_track_ids = [track['id'] for track in library
                   if track['artist'] == 'The Cat Empire']
```
from: https://github.com/simon-weber/gmusicapi

closed 2018-05-28 by leavmealonepy

shivasiddharth, 2018-05-29:
I shall add a script to generate the device-id, so that ppl can need not go back and forth editing it in the actions.py script.