shivasiddharth/GassistPi #365: Magic Mirror integration

bony4u, 2018-05-29:
boss, i am having issue when using Magic Mirror commands, actually, it is not working, can you explain the procedure to add the remote control module to gassist pi?

shivasiddharth, 2018-05-30:
Procedure has clearly been given in README and already there are a number of issues, where i have discussed this.

closed 2018-05-30 by shivasiddharth

MMZCustoms, 2018-05-30:
I know, but i tried all. And done everything as described in readme. 
Remote control through web is working. But through voice not working. In Gassist Pi, i just need to replace mmmip '<ip>' in main.py right?.
Actually am using same pi for both. Means both MagicMirror and Gassist Pi in same pi. 
So i type the ip of that pi.

shivasiddharth, 2018-05-30:
Please refer to #352

MMZCustoms, 2018-05-30:
Actually the problem is with magic mirror turn display off command. I fixed it. Thank you for your support. 
And one more thing, when i use trigger shutdown command it is showing some search results on my phone.