shivasiddharth/GassistPi #367: modelid.txt not generated

codelock13, 2018-05-31:
what i have to enter in "Enter the modelid that was generated in the actions console:"?
modelid.txt is not found in /home/pi/ directory and also do i need to enter path of credentials file??
for eg. /home/pi/*********.json where ********* is name of .json file?

aandroide, 2018-05-31:
sorry but have you read the guide?
the model id and the project id is generated when you register the device in the action console
1.Follow the instructions here to Configure a Developer Project and Account Settings. Then follow this guide to register the device and obtain the credentials.json file.

2.Place the credentials.json file in/home/pi directory DO NOT RENAME

codelock13, 2018-06-01:
Can you plz link the guide the readme.MD doesn't show everything.

codelock13, 2018-06-03:
I got it but encountered another glitch at the end of installation when I get URL to authenciate then suddenly many errors appear an on terminal two new command get executed they are device type and error and installation don't get executed .

2.0; extra == "samples"->google-assistant-sdk[samples]==0.5.0) (39.2.0)
Please visit this URL to authorize this application: https://accounts.google.com/o/oauth2/auth?response_type=code&client_id=447022059209-kujvvk62pck2tjjdrhq8i2jqj98ce7f9.apps.googleusercontent.com&redirect_uri=urn%3Aietf%3Awg%3Aoauth%3A2.0%3Aoob&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype+https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fgcm&state=Q6RHBRn2hA04lI2Iwi1o8zg9w7NYcF&access_type=offline&prompt=consent
Enter the authorization code: Traceback (most recent call last):
  File "/home/pi/env/bin/google-oauthlib-tool", line 11, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/tool/__main__.py", line 106, in main
    creds = flow.run_console()
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/flow.py", line 364, in run_console
    self.fetch_token(code=code)
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/flow.py", line 241, in fetch_token
    self.client_config['token_uri'], **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests_oauthlib/oauth2_session.py", line 244, in fetch_token
    self._client.parse_request_body_response(r.text, scope=self.scope)
  File "/home/pi/env/lib/python3.5/site-packages/oauthlib/oauth2/rfc6749/clients/base.py", line 411, in parse_request_body_response
    self.token = parse_token_response(body, scope=scope)
  File "/home/pi/env/lib/python3.5/site-packages/oauthlib/oauth2/rfc6749/parameters.py", line 379, in parse_token_response
    validate_token_parameters(params)
  File "/home/pi/env/lib/python3.5/site-packages/oauthlib/oauth2/rfc6749/parameters.py", line 386, in validate_token_parameters
    raise_from_error(params.get('error'), params)
  File "/home/pi/env/lib/python3.5/site-packages/oauthlib/oauth2/rfc6749/errors.py", line 415, in raise_from_error
    raise cls(**kwargs)
oauthlib.oauth2.rfc6749.errors.InvalidGrantError: (invalid_grant) Malformed auth code.
pi@raspberrypi:~ $ Auto
bash: Auto: command not found
pi@raspberrypi:~ $ 

This is how it looks like plz help

aandroide, 2018-06-03:
Hai visitato la url che ti viene mostrata?

codelock13, 2018-06-04:
I cannot understand Italian.

codelock13, 2018-06-04:
And I have not gone to that URL as before I get time to copy URL.the script end as I described executing two commands device type and auto both of them get command not found automatically.

aandroide, 2018-06-04:
I hope this can help you even if it is an old video and no longer working that way but about the authorization of the token through the url, it remained the same.
watch from 12:35 minute
https://www.youtube.com/watch?v=vprJBDDYE8Q

shivasiddharth, 2018-06-04:
@codelock13 , I had to delete that comment, because it had the authorization URL. 
I see the problem that you are facing. I guess you are copying URL by selecting everything and using right click copy. 

When inside the terminal, if you make a selection and use right click, it will paste the content you copied in the next line and that is what is happening in your case. If you move the mouse pointer over the URL, you should get an option to copy the link. Use that to copy the link and authenticate.

Modelid is no longer generated by the script. You will be provided with the model id in the actions console while registering. All the links have been provided in the README. The links are hyperlinked in the text.

closed 2018-06-04 by shivasiddharth