shivasiddharth/GassistPi #368: FATAL:audio_input_stream.cc

TamojitSaha, 2018-05-31:
### Describe the bug:
```
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
ALSA lib pcm_direct.c:1605:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
/home/pi/GassistPi/scripts/gassist-installer-pi3.sh: line 63:  5166 Aborted                 googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
I think so

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
ALSA lib pcm_direct.c:1605:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
/home/pi/GassistPi/scripts/gassist-installer-pi3.sh: line 63:  5166 Aborted                 googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid
```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Not yet tested at the time of reporting the current issue.

aandroide, 2018-05-31:
you can publish your files  .asoundrc and asound.conf

TamojitSaha, 2018-05-31:
Can this happen without plugging of usb soundcard??

aandroide, 2018-05-31:
certainly. What do you use as a microphone and as a speaker?

TamojitSaha, 2018-05-31:
I didn't plugged the usb sound card for microphone input. I tested out speaker and it worked out fine.
Also can you just guide me on how to incorporate custom voice commands manually to operate gpio.

Suppose I say **Hey Google, shut down garage doors**
It should output low or high logic at the assigned gpio .

aandroide, 2018-05-31:
with regard to the relays, they are not all the same, some may close the contact using high logic and others with low logic. you must always look at the datasheet of the component you buy.
That said, what you ask for is very simple given that the gassistpi project already uses the gpio library.
Example: if the relay connects it to gpio18, just state it in the main.py file GPIO.setup (18, GPIO.OUT, initial = GPIO.LOW)
to activate the relay you will have to use GPIO.output (18, GPIO.HIGH)
and to turn it off GPIO.output (18, GPIO.LOW)
so to integrate it with the assistant you have to add this command:
if 'shut down garage doors'.lower () in str (usrcmd) .lower ():
assistant.stop_conversation ()
GPIO.output (18, GPIO.HIGH)
say ("I close the garage door")
obviously this is just an example
if you want to do something more organized, create the action in the actions.py file and then call it in main.py as it is done for radio stations and other functions.

shivasiddharth, 2018-06-02:
Without plugging the mic, one cannot expect the assistant to work. Closing this..

closed 2018-06-02 by shivasiddharth