shivasiddharth/GassistPi #378: "Give me the permissions to help you"

MBenoli, 2018-06-03:
Hi,

The install worked successfully and the Google Assistant answers. Thank you for that.

However, when I try to ask it personal stuff (e.g. "where do I work ? or "can you remember..."), it keeps telling me: "Give me the permissions to help you". Or, since I don't have an android device, I cannot access a mobile device and setup the permissions. So, I would like to give the permissions directly from the Raspberry Pi. Of course, I have done this: https://stackoverflow.com/questions/46482807/google-assistant-sdk-returns-give-me-the-permission-to-help-you

But it is not sufficient. Any idea, on how to solve this issue ?

Also, any idea on how to change the language ? Tried the common command existing for android devices but it does not work.

Thanks !

kodxana, 2018-06-04:
You need use Google Assistant App , try borow phone from friend for example.

MBenoli, 2018-06-05:
Yep. Did that and it works. Thanks.

closed 2018-06-05 by MBenoli