shivasiddharth/GassistPi #381: Music Volume is not saving

leavmealonepy, 2018-06-05:
Command ,,Hey Google, set music volume to maximum"
works only once, after reboot or changing source of stream from for example. from google music to youtube music is muted and I need to say command again.

shivasiddharth, 2018-06-05:
Can you add this line `self.check_delete("/home/pi/.mediavolume.json")` under` def stop_vlc(self):` in the mediaplayer.py file and then try.

leavmealonepy, 2018-06-05:
like
```
    def stop_vlc(self):
	self.check_delete("/home/pi/.mediavolume.json")
        print('stopping vlc')
        self.libvlc_player.stop()
```
?

leavmealonepy, 2018-06-05:
Assistant wont boot after this change @UP

shivasiddharth, 2018-06-05:
If it was entered correctly, there should not be any issues. Manually start the assistant and post the error.

leavmealonepy, 2018-06-05:
Ok, I have fixed it, it was Python Syntax error with tabing and spacing.
Now after manual start it shows:
```
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'pi-home-a4520-google-pi-home-lq8tmd'
[028f3058] pulse audio output error: PulseAudio server connection failure: Connection refused
[0292e8f0] pulse audio output error: PulseAudio server connection failure: Connection refused
device_model_id: pi-home-a4520-google-pi-home-lq8tmd
device_id: 9086AE97C3552300715A398856BC3792

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
```
I haven't tested that its fixed yet, but isn't PulseAudio a reason of problems??

shivasiddharth, 2018-06-05:
No. This is usual, nothing to worry.

leavmealonepy, 2018-06-05:
Google music volume was set to maximum after reboot but youtube was still muted.
After I have changed back from youtube to google music, it was muted too.

And I dont know why, maybe its my accent fault, but usually Hey Google Pause Music worked, now it won't propably after this change.

leavmealonepy, 2018-06-05:
```
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "stop music"}
sh: 1: /home/pi/.mediavolume.json: Permission denied
stopping vlc
ON_RENDER_RESPONSE:
  {"text": "Nothing's playing at the moment.", "type": 0}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
```
Okay, now I see its Permission denied, the fun  fact is that I am running it on root access account :V

leavmealonepy, 2018-06-05:
started with sudo:
```
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "stream Arctic Monkeys balaclava"}
sh: 1: /home/pi/.mediavolume.json: Permission denied
stopping vlc
Getting youtube link
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
```

and Google Music:

```
Artist not found
Artist not found
Artist not found
Artist not found
sh: 1: /home/pi/.player.json: Permission denied
sh: 1: /home/pi/.trackqueue.json: Permission denied
[014d8410] pulse audio output error: PulseAudio server connection failure: Connection refused
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
```
Still Permission denied lol

leavmealonepy, 2018-06-05:
No permission denied now, but music volume is still not saving

shivasiddharth, 2018-06-05:
I have added a bug tag, will try and fix it in the next couple of days

shivasiddharth, 2018-06-05:
I have fixed the issue, you need not modify much, just check the commit and add those few lines to your existing script.

closed 2018-06-05 by shivasiddharth

leavmealonepy, 2018-06-05:
Pulled from github and reedited my login details, I can confirm - fix is working.
Good job like always! Thanks!

btw. Your response time is brilliant

Akshaykkapoor, 2018-06-07:
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play all songs from Google Music"}
stopping vlc
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

 
still getting the stopping vlc as response , pulled the code from git today ..

commit c8bce5af (2018-06-05): Fixed Music streaming volume bug

Fixes #381