shivasiddharth/GassistPi #382: Issue with 'Hey Google Good Morning' or O'k Google Good Night' commands

amitsrivastava79, 2018-06-06:
### Describe the bug:
Hi Sid,

**Thanks for all the good job** you did, especially by creating automated Gassist, with this it's hardly an hour work to run the assistant on the PI... manual setup wasted my half day and still not all feature like Youtube stream was there.

Yesterday Night instead of my manual setup of Google Home on Pi, I used Gassist automated scripts. 

**My observations and where need your support:**

**1.** After the Manual setup as per Google docs, when I use commands Like 'Hey Google Good Morning' or Ok Google Good Night... it responds me very well like in morning it briefs me about the weather and then how should I reach office needed time to reach and road congestions, also news breifings. The same case with Good Night command ..it responds sleep tight and when you want to wake up and set the alarms.  

>>>> But with Gassist, both the commands are not working, it listens to the commands and then seems PI got busy GPIo bulbs keep blinking, but no response. 

Am I missing something or these commands are changed in the scripts, I tried but not able to find such updates. Please Suggest

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-06-06:
Without the contents of the terminal or the log file, there is nothing much that i can do.

amitsrivastava79, 2018-06-06:
Hi Sid,

Thanks for quick response and support.  I am not good in programming and technical details so might be it's a dumb question but I hope with log file you mean gassistpi.log in temp .

If yes, I will send it tonight else let me know the log file and it's location.

amitsrivastava79, 2018-06-06:
Hi Sid,

I updated the terminal output as well as log file output, during command 'Good Morning' there was some exception as in log file but for 'Good Night' no exception thrown.

One More thing I want to highlight in terminal it showing different question like "Ok, let's get ready for bed" and "Alarm" , but in actual, there was no voice from the speaker(connected to PI on 3.5 Jack). although speakers are perfectly connected and voice is there.

Second thing, during Good Morning command it also brief the news to my surprise it's actually playing a video but on my chromecast device but no voice and other details in speaker.

I also verify the Stream command and found with stream command it's actually casting the video on my chromecast, also there is voice on the spk connected to Pi 3.5 jack but voice quality was bad.

Hope you get some insight on this issue and help me out.

shivasiddharth, 2018-06-06:
> I also verify the Stream command and found with stream command it's actually casting the video on my chromecast, also there is voice on the spk connected to Pi 3.5 jack but voice quality was bad.

Since you have already posted this in another thread, i shall not address this here. "One place for one error."

1. Good Morning issue:
I have added a custom command for reading out news from RSS feeds. Your news routine is triggering that. Either change the word news to something else or comment out the custom action [here](https://github.com/shivasiddharth/GassistPi/blob/c8bce5afd9cc36f5f239abe3f8995bb3e61323b4/src/main.py#L406).

2. Good Night:
Did you stop the service, before manually running the assistant ?

amitsrivastava79, 2018-06-07:
Hi Sid,

Thanks for your quick response and support, I will try to comment or change the word news for Good Morning Issue. 

For Good Night issue, I haven't stopped the service manually.  
Yesterday Night I did some changes, also power off the chrome cast and after that I was able to get some output for Good Night command ...although there is double sound for all question like ECHo, I will get back to you with finer details on the issue ... Thanks Again :)

amitsrivastava79, 2018-06-08:
Hi Sid,

I tried by commenting the lines with news word as mentioned by you, but still same. I tried to dig more and my findings are except 'stream' output all audio output is diverting to my chrome cast instead of 3.5 spk connected to PI. (Although I haven't set it's IP in the Action.py)

In the attached assistant output you can see, with every command google responded with some text, but same is not audible to me on the PI Spk.. except the stream command. 

Even when I switched off my chromecast it's diverting voice to it later it throws an error on assistant output as device switched off.  

Later I logged in my google assistant app and deleted my chromecast and rerun some command still you can see in the attached assistant output it throws error as 'you'll need a Chromecast or cast-enabled TV linked to your account' So basically it's not giving me any answer via voice although assistant output showing it reverting.
[Assistant.txt](https://github.com/shivasiddharth/GassistPi/files/2085829/Assistant.txt)
[Logs.txt](https://github.com/shivasiddharth/GassistPi/files/2085830/Logs.txt)

For your reference, I attached the log and assistant file output. Please help

shivasiddharth, 2018-06-08:
Regarding the stream command issue: Have you modified anything in the main.py script?

Frankly, this is the first time i am coming across someone saying that assistant's response was being played through another another device. It could be some settings in your home app, but defenitely it has nothing to do with the Pi scripts.

amitsrivastava79, 2018-06-09:
Ok, but I haven't modified anything in your scripts or main.py.
In Google home App, my chromecast was added, but that's not the issue as even when I removed it ...the assistant output keeps saying "you'll need a Chromecast or cast-enabled TV linked to your account", so no voice from my spkrs connected to PI.

I understand that it's a strange issue, I will try to reinstall all from scratch and let see if I found any hope /update on my issue

amitsrivastava79, 2018-06-10:
One more finding after diagnoses, the output of normal commands are also there now in 3.5 MM Spk jack but that so low you can't hear it, the volume of spk as well as via command already set to maximum, but still not audible. 

I tried changing the power supply, instead of local spk used my bose but nothing worked and the issue is that voice is only for normal queries via assistant like general talk, but for other things like Stream music , PI boot up sound, every sound is too loud and audible. 

Looks like some audio quality rate issue, is it that the stream is giving 44000hz quality but the normal audio is on 16000Hz? can you guide where I can change the audio quality to give it a try.

shivasiddharth, 2018-06-10:
I can only suggest you to use a USB sound card or an I2S DAC