shivasiddharth/GassistPi #383: YouTube streaming stoped working?

leavmealonepy, 2018-06-07:
Are there any changes in YT API?
Stoped working just like that.
```
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0
2018-06-07 17:15:22,146 INFO googleapiclient.discovery URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest
2018-06-07 17:15:22,473 INFO googleapiclient.discovery URL being requested: GET https://www.googleapis.com/youtube/v3/search?q=the+killers+the+man&part=id%2Csnippet&alt=json&key=AIza(HIDDEN KEY)
2018-06-07 17:15:25,968 ERROR __main__ argument 2: <class 'TypeError'>: wrong type
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 508, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 394, in main
    YouTube_No_Autoplay(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1035, in YouTube_No_Autoplay
    vlcplayer.youtube_player(currenttrackid)
  File "/home/pi/GassistPi/src/mediaplayer.py", line 118, in youtube_player
    self.media_player(tracks[trackid])
  File "/home/pi/GassistPi/src/mediaplayer.py", line 58, in media_player
    self.set_vlc_volume(setvollevel)
  File "/home/pi/GassistPi/src/mediaplayer.py", line 67, in set_vlc_volume
    self.libvlc_player.audio_set_volume(level)
  File "/home/pi/env/lib/python3.5/site-packages/vlc.py", line 4247, in audio_set_volume
    return libvlc_audio_set_volume(self, i_volume)
  File "/home/pi/env/lib/python3.5/site-packages/vlc.py", line 7682, in libvlc_audio_set_volume
    return f(p_mi, i_volume)
```

shivasiddharth, 2018-06-07:
`GET https://www.googleapis.com/youtube/v3/search?q=the+killers+the+man&part=id%2Csnippet&alt=json&key=AIza(HIDDEN KEY)`

 Nothing has changed with the API, most likely you would have used up the free quota. Paste that url in your web browser and revert back with the response that you get. (URL to be pasted with your full credentials).

leavmealonepy, 2018-06-07:
[return.txt](https://github.com/shivasiddharth/GassistPi/files/2081181/return.txt)

shivasiddharth, 2018-06-07:
Is the google music running properly ?

leavmealonepy, 2018-06-07:
GMusic is restarting my device too
```
2018-06-07 20:49:01,865 DEBUG gmusicapi.Mobileclient1 GetStreamUrl(args=['6aa4e56a-8b83-38b0-b7c8-7ee6fc724058', '4540056832825989280', 'hi'], kwargs={})
2018-06-07 20:49:01,872 DEBUG urllib3.connectionpool Resetting dropped connection: mclients.googleapis.com
2018-06-07 20:49:02,895 DEBUG urllib3.connectionpool https://mclients.googleapis.com:443 "GET /music/mplay?slt=1528397341866&hl=en_US&songid=6aa4e56a-8b83-38b0-b7c8-7ee6fc724058&pt=e&net=mob&sig=vzbyJaHvVzoA6OZ908oYxf8FKTo&opt=hi&dv=0&tier=aa HTTP/1.1" 302 None
2018-06-07 20:49:02,977 DEBUG gmusicapi.Mobileclient1 https://r4---sn-5hne6nsy.c.doc-0-0-sj.sj.googleusercontent.com/videoplayback?id=be10c92fe19e8ec4&itag=25&source=skyjam&begin=0&ei=Hn4ZW6jpHo-b_gaes6W4BQ&o=15734057790582887340&cmbypass=yes&ratebypass=yes&cpn=SoG4DYaz3oOLmHhNbO0c4Q&ip<MYIPWASTHERE>&ipbits=0&expire=1528397432&sparams=cmbypass,ei,expire,id,initcwndbps,ip,ipbits,itag,mm,mn,ms,mv,o,pl,ratebypass,source&signature=497D0BD9989B680EADF74D927636BDFF9471B2E7.80DF27D8A96865F585DCDABE00961395C364B71D&key=cms1&initcwndbps=5010&mm=31&mn=sn-5hne6nsy&ms=au&mt=1528397249&mv=m&pl=21
2018-06-07 20:49:03,221 ERROR __main__ argument 2: <class 'TypeError'>: wrong type
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 508, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 503, in main
    gmusicselect(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 985, in gmusicselect
    vlcplayer.googlemusic_player(currenttrackid)
  File "/home/pi/GassistPi/src/mediaplayer.py", line 112, in googlemusic_player
    self.media_player(streamurl)
  File "/home/pi/GassistPi/src/mediaplayer.py", line 58, in media_player
    self.set_vlc_volume(setvollevel)
  File "/home/pi/GassistPi/src/mediaplayer.py", line 67, in set_vlc_volume
    self.libvlc_player.audio_set_volume(level)
  File "/home/pi/env/lib/python3.5/site-packages/vlc.py", line 4247, in audio_set_volume
    return libvlc_audio_set_volume(self, i_volume)
  File "/home/pi/env/lib/python3.5/site-packages/vlc.py", line 7682, in libvlc_audio_set_volume
    return f(p_mi, i_volume)
ctypes.ArgumentError: argument 2: <class 'TypeError'>: wrong type
2018-06-07 20:49:14,114 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-06-07 20:49:14,122 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-06-07 20:49:14,765 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
```

shivasiddharth, 2018-06-07:
Its a VLC issue then. Will check is out over the weekend.

leavmealonepy, 2018-06-07:
It just stopped working just like that.
It worked today and 2 hours later it stopped working.

shivasiddharth, 2018-06-07:
Can you try commenting the lines that i added few days back in the mediaplayer.py. under the media_player function to control the volume.

leavmealonepy, 2018-06-07:
which ones?
Or you just want me to put old script back?

shivasiddharth, 2018-06-07:
From the log file, i see that there is some issue in setting the vlc volume, so please comment [these](https://github.com/shivasiddharth/GassistPi/blob/c8bce5afd9cc36f5f239abe3f8995bb3e61323b4/src/mediaplayer.py#L55) 55 to 60 lines out and then try.

leavmealonepy, 2018-06-07:
It works after commenting lines |55->60

leavmealonepy, 2018-06-07:
It propably stoped working after I said something like ,,Hey Google, set music volume to 50" today.

commit 483e8301 (2018-06-12): Update mediaplayer.py

Fixes #389 Fixes #383.
The volume levels are stored as a string and hence resulting in the crash of the assistant when VLC attempts to set the volume. This fix converts string to integer, thus preventing the crash.
closed 2018-06-12 by shivasiddharth