shivasiddharth/GassistPi #384: Only English

Oscaretto, 2018-06-07:
Hello, i have installed GassistPi and it worked very well also in italian, but when i have changed the address where is located "Google Home" from the google assistant settings from android the language has changed in english... from the settings the italian language there isn't. I must reinstall GassistPi o there is a solution?
Thanks to someone who help me

shivasiddharth, 2018-06-07:
@aandroide , May be you can help..

aandroide, 2018-06-07:
Ciao che versione di Android Usi? Devi usare l'applicazione assistant sul telefono andare in impostazioni, cercare il tuo dispositivo e cambiare la lingua. L'italiano è stato introdotto a marzo 2018 controlla se devi aggiornare l'applicazione

Oscaretto, 2018-06-08:
Utilizzo Android 7.1.1 e nell'app escono tutte le lingue tranne che italiano

aandroide, 2018-06-08:
Ciao ho controllato ed effettivamente è stata rimossa la lingua italiana.  non dipende dal progetto . Sicuramente google sta rilasciando un nuovo aggiornamento lato server. Attendiamo qualche giorno.

Oscaretto, 2018-06-11:
Quindi?

aandroide, 2018-06-11:
hai già provato a impostare l'indirizzo del dispositivo nell'app assistant?
Sinceramente non ho idea del perchè hanno rimosso la lingua italiana in hotworld.
Faccio una segnalazione

Oscaretto, 2018-06-12:
Proprio quando ho cambiato l'indirizzo da italiano è diventato inglese

aandroide, 2018-06-12:
altre persone hanno già segnalato il problema ma nessuno per adesso ha ancora risposto.
Segui la discussione di spechhio magico qui: https://plus.google.com/communities/117537996116836200696

aandroide, 2018-06-19:
@Oscaretto ho fatto la segnalazione qui:
https://github.com/googlesamples/assistant-sdk-python/issues/259

aandroide, 2018-07-11:
@Oscaretto la lingia italiana è disponibile se usi un dispositivo ios. accedi al tuo account sull'app assistant su iphone è l'unico metodo funzionante per ora.
Su android sono troppo lenti a risolvere i problemi.
Ciao

NeoMod, 2018-07-12:
Grazie per la segnalazione @aandroide 

Trovo assurdo, anzi ai limiti del ridicolo paradassole, che abbiano risolto su iOS e non su Android...

shivasiddharth, 2018-07-18:
Whats the status of this issue ? Can it be closed?

NeoMod, 2018-07-18:
@shivasiddharth Technically it is not an issue strictly related to GassistPI, but I think it would be useful to leave this topic open to serve as a "landing point" for whomever has the same problem.

This way as soon as Google solves the problem also on Android devices one could inform users via this topic and then close it.

(I believe many users will easily find this discussion instead of the official google thread).

Supermene, 2018-07-18:
I'm currently doing this project and I also see that Italian language is still unavailable on GassistPi but meanwhile on android smartphone is available. Anyway I sent a feedback to developers, hoping they will resolve it soon!

NeoMod, 2018-07-18:
Yeah, unfortunately they still have to solve the issue on Android.

Anyway, as a temporary solution as mentioned by aandroide, if you have access to an Apple device you can log-in there via the Assistant App and then select the Italian Language for your GassistPi-device.

fragolinux, 2018-07-29:
i've set italian language on android google home app, but gassistpi continues to use english...

aandroide, 2018-07-29:
@fragolinux non devi usare l'app di google home ma bensi' l'assistente google entrare in impostazioni, selezioni il tuo dispositivo e poi vai su lingua.
Mi dispiace informarti che ad oggi su android non hanno ancora risolto il problema di rimettere la lingua italiana tra le scelte disponibili, e quindi l'unico metodo rimane ancora quello di usare un iphone ed accedere con il tuo account google, apri l'app google assistant e segui i passaggi scritti sopra.

fragolinux, 2018-07-29:
ho rifatto il setup usando la versione con desktop pixel, e usando l'app google home sono riuscito a mettere il raspi in italiano... se prima gli dicevo "qual è la capitale d'italia" lui cercava di interpretare in inglese, ora in console vedo chiaramente che riconosce tutto... il problema ora resta che il raspi non parla, pare per un problema o di lentezza di banda o di superamento dei 500 token quotidiani... mi pare ben strano, sono su una 100mb e di sicuro non ho usato 500 token oggi... info qui: https://github.com/shivasiddharth/GassistPi/issues/462

NeoMod, 2018-07-29:
Come si segnala qui (https://github.com/googlesamples/assistant-sdk-python/issues/274) il nuovo problema in cui sei incorso non riguarda la localizzazione bensì una errata gestione da parte di Google della procedura di autenticazione.

Sia per il sistema "push-to-talk" che per quello "voice-actrivated" viene restituito, in fase di prima configurazione, un prompt HTML al quale l'utente dovrebbe accedere per consentire a Google Assistant l'accesso al proprio account.

Purtroppo l'HTML generato appare essere una pagina statica che non conduce effettivamente a nessuna autorizzazione (quando visualizzata in ambiente desktop durante la configurazione push-to-talk usando il flag --display).

aandroide, 2018-08-03:
https://plus.google.com/u/0/106053454660110517628/posts/SbCZFcnvvCn?cfem=1
È stata ripristinata la lingua italiana su android

aro28, 2018-08-17:
Hi All,
Google Assistant SDK for devices.
The Google Assistant Library can recognize speech (hotword models and queries) and respond in the following languages and locales:

Language | Language-Region | Language (English name)
-- | -- | --
Deutsch (Deutschland) | de-DE | German (Germany)
English (Australia) | en-AU | English (Australia)
English (Canada) | en-CA | English (Canada)
English (Great Britain) | en-GB | English (United Kingdom)
English (United States) | en-US | English (United States)
Français (Canada) | fr-CA | French (Canada)
Français (France) | fr-FR | French (France)
Italiano (Italia) | it-IT | Italian (Italy)
日本語（日本） | ja-JP | Japanese (Japan)

https://developers.google.com/assistant/sdk/reference/library/languages

shivasiddharth, 2018-11-30:
Do we have  a status on this issue ?

shivasiddharth, 2018-11-30:
Apparently, some languages work only on phones https://support.google.com/assistant/answer/7394513?co=GENIE.Platform%3DAndroid&hl=en It might be confusing for a newcomer.

shivasiddharth, 2019-01-16:
Closing this...

closed 2019-01-16 by shivasiddharth