shivasiddharth/GassistPi #385: No Fb.wav voice after saying "ok, Google"

aro28, 2018-06-07:
Hi! When i say "Ok, Google" it should play Fb.wav voice from /home/pi/GassistPi/sample-audio-files directory. But now, it works without beeping. Who know how can i fix it.

Thanks in advance.

shivasiddharth, 2018-06-07:
If you are using the demo, you will not hear the beep. For better understanding, it would be nice to mention the board that you are using along with the audio peripherals. As it is, the issue that you have opened conveys very little info.

aro28, 2018-06-07:
As I know I do not use Demo. How can I check it?
I use USB mic and USB sound adapter. 
I configured below audio settings. 
3.4. USB MIC AND AUDIO JACK users,
sudo chmod +x /home/pi/GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh  
sudo /home/pi/GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh  

speaker-test -t wav command is passed successfully.

shivasiddharth, 2018-06-07:
Frankly, i have no idea as to why that is happening. All that i can say is, if you are using USB Mic and USB sound card, you have got the configuration scripts wrong. The scripts that you have used are for Pi's onboard jack.

aro28, 2018-06-07:
Ok. Will try to setup with onboard audio tomorrow. 
Thanks!

aro28, 2018-06-08:
HI @shivasiddharth 
I reconfigured for USB mic and onboard audio jack. It's still there's no confirmation beep.

aro28, 2018-06-08:
Fixed. It was an audio settings issue. :)

closed 2018-06-08 by shivasiddharth