shivasiddharth/GassistPi #386: HEADLESS AUTOSTART / BOOT SERVICE SETUP problem

snowythegreat, 2018-06-08:
## **My Hardware configuration .......**
Raspberry Pi 2B, OS- Raspbian Stretch (latest version), USB mic and audio 
thru audio socket.

Issue .......
GassistPi working fine with manual star using this command...
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'replace this with the model id'

## **However when trying HEADLESS AUTOSTART on BOOT SERVICE SETUP ......**

pi@raspberrypi:~ $ sudo systemctl enable gassistpi-ok-google.service  
Failed to enable unit: File gassistpi-ok-google.service: Invalid argument
pi@raspberrypi:~ $ 

#########################################################################

## **ON RUNNING systemctl status ....**

pi@raspberrypi:~ $ systemctl status gassistpi-ok-google.service
● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)

Jun 08 12:12:12 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:9] Unbalanced quoting, ignoring: "/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'gassistshiva07062
Jun 08 12:12:12 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:10] Missing '='.
Jun 08 12:20:31 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:9] Unbalanced quoting, ignoring: "/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'gassistshiva07062
Jun 08 12:20:31 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:10] Missing '='.
Jun 08 12:20:48 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:9] Unbalanced quoting, ignoring: "/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'gassistshiva07062
Jun 08 12:20:48 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:10] Missing '='.
Jun 08 12:24:08 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:9] Unbalanced quoting, ignoring: "/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'gassistshiva07062
Jun 08 12:24:08 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:10] Missing '='.
Jun 08 12:24:28 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:9] Unbalanced quoting, ignoring: "/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'gassistshiva07062
Jun 08 12:24:28 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:10] Missing '='.
lines 1-14/14 (END)
################################################

## **My gassistpi-ok-google.service file ......**

[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'gassistshiva07062018-mygassistshiva-xcaoqh'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

##################################################################
Appears to be a syntax error but I couldn´t catch it..

shivasiddharth, 2018-06-08:
> ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'gassistshiva07062018-mygassistshiva-xcaoqh'

The model id is on the second line.  I am not sure if it is github's formating or whether that is how it is in your file.

snowythegreat, 2018-06-08:
This is github's formating .... in the file the whole sentence is in one line.

snowythegreat, 2018-06-08:
@shivasiddharth  ...... Notwithstanding these minor issues, You´ve done an excellent service to enthusiasts by simplifying the Google Assistant installation process ..... I could get it up and running in the first attempt .... Not to mention the tons of additional features .....Thanks a ton

shivasiddharth, 2018-06-08:
``--device_model_id 'gassistshiva07062018-mygassistshiva-xcaoqh' can you check the space before the quotes. I think it needs a double spacing.

snowythegreat, 2018-06-09:
@shivasiddharth  ...... I have found the source of the problem ......

The device_model_id should be **_WITHOUT_** quotes ...
device_model_id gassistshiva07062018-mygassistshiva-xcaoqh
(Please refer your Youtube tutorial....( 08.43 min onwards ..) **How to Install the New Google Assistant SDK on Pi and Set it up as a Headless unit to Auto Start** @ https://www.youtube.com/watch?v=vprJBDDYE8Q&t=559s )

With that amendment, the autostart is working like a charm ......
You may like to emphasis this in your instructions.....

Thanks a lot .... Closing the issue.

closed 2018-06-09 by snowythegreat