shivasiddharth/GassistPi #387: Philips Hue integration

aro28, 2018-06-09:
Hi everyone!
I’m wondering how to integrate my Hue controller and bulbs with Google Assistant. I wish to turn on/off Hue bulbs by voice commands. 
Does anybody configured successfully on MM?
Below step is not clear for me. 
- Download sketches for your NodeMCU/Wemos/ESP Devices from here

aro28, 2018-06-09:
Step 19 is not clear for me. :)

shivasiddharth, 2018-06-09:
pls use gitter for non-issue posts.

closed 2018-06-09 by shivasiddharth

aro28, 2018-06-10:
Hi @shivasiddharth 
I'm sorry, but i'm a new user on Github and never used gitter IM platform. Will start to work with Gitter.
At this moment i have Philips Hue Smart bridge with Smart bulbs and it works and i can remote them thru my iPhone device.
I've read Read.md file on "EMULATED PHILIPS HUE SERVICE AND CONTROL" section (step 19) and i can not understand clearly how to setup it.

Should i use NodeMCU devices for controlling my Hue lights? I never used and work with that type of devices. If yes, i would be appreciate if you could me please send me a proper link for guideline.
Many thanks for your great work and efforts.
Best,
Ruslan