shivasiddharth/GassistPi #388: ALSA lib pcm.c:8389:(snd_pcm_set_params) Channels count (2) not available for CAPTURE: Invalid argument

Aftabx, 2018-06-09:
### Describe the bug:
```

Dear, 
During the execution of the script :  sudo ./gassist-installer-pi3.sh 
I am getting an error 
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```
> 

credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
ALSA lib pcm.c:8389:(snd_pcm_set_params) Channels count (2) not available for CAPTURE: Invalid argument
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
./gassist-installer-pi3.sh: line 54:  3320 Aborted                 googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid
pi@raspberrypi:~/GassistPi/scripts $ 

> 
```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-06-10:
looks like your ic was not plugged in while running the setup.

Aftabx, 2018-06-10:
Is there some other command that i can run, to help to debug the problem? :)

Thanks in advance for your help

shivasiddharth, 2018-06-10:
I need some more info like the audio configuration that you chose to setup. Also the output of the following:
```
aplay -l
arecord -l
```

Aftabx, 2018-06-10:
My audio configuration :

> pi@raspberrypi:~ $ sudo raspi-config 
numid=3,iface=MIXER,name='CH3 digital volume'
  ; type=INTEGER,access=rw---R--,values=1,min=0,max=255,step=0
  : values=1
  | dBscale-min=-119.25dB,step=0.75dB,mute=0
pi@raspberrypi:~ $ 

> 

``

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

``

`pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed4micvoicec [seeed-4mic-voicecard], device 0: bcm2835-i2s-ac10x-codec0 ac10x-codec0-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ `

shivasiddharth, 2018-06-10:
Can you also paste the output of 
```
sudo nano /etc/asound.conf
sudo nano ~/.asoundrc
```

Aftabx, 2018-06-10:
amazing,

When i have  run the nano on ~/.asoundrc, i have noticed that the config was no more the one i setup.
Luckily i had a backup of my config.
So i have manage to get the correct config of : ~/.asoundrc

then suddenly i started the hear audio from my jack device. 
And the google assistant was working.

Even after the reboot of my Raspberry the google assistant is working.

Thanks a lot for your support.

closed 2018-06-10 by shivasiddharth

bachoo786, 2019-03-02:
@Aftabx  can you please share your /etc/asound.conf and ~/.asoundrc please I am having the same problem as you.

thanks

subham9777, 2020-02-22:
Hey @shivasiddharth @Aftabx Same problems is still happening and i did a clean install and all .
Here are my files

arecord -L

null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
multiapps
dmixer
default
output
usbstream:CARD=ALSA
    bcm2835 ALSA
    USB Stream Output
sysdefault:CARD=seeed8micvoicec
    seeed-8mic-voicecard, 
    Default Audio Device
dmix:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample mixing device
dsnoop:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample snooping device
hw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct hardware device without any conversions
plughw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Hardware device with all software conversions
usbstream:CARD=seeed8micvoicec
    seeed-8mic-voicecard
    USB Stream Output

aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
multiapps
dmixer
default
output
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dmix:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Hardware device with all software conversions
usbstream:CARD=ALSA
    bcm2835 ALSA
    USB Stream Output
sysdefault:CARD=seeed8micvoicec
    seeed-8mic-voicecard, 
    Default Audio Device
dmix:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample mixing device
dsnoop:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample snooping device
hw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct hardware device without any conversions
plughw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Hardware device with all software conversions
usbstream:CARD=seeed8micvoicec
    seeed-8mic-voicecard
    USB Stream Output

sudo nano /etc/asound.conf
# The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one

# use samplerate to resample as speexdsp resample is broken
defaults.pcm.rate_converter "samplerate"

pcm.!default {
    type asym
    playback.pcm "dmixer"
    capture.pcm "multiapps"
}

pcm.multiapps {
     type plug
     slave.pcm {
         type dsnoop
         slave {
             rate 48000
             format S32_LE
             pcm "hw:seeed8micvoicec"
         }
         ipc_key 666666
     }
 }

pcm.dmixer {
    type plug
    slave {
        pcm {
            type dmix
            ipc_key 555555
            slave {	
                pcm "hw:seeed8micvoicec"
                format S32_LE
                channels 8
            }
            bindings {
                0 0
                1 1
                2 2
                3 3
                4 4
                5 5
                6 6
                7 7
            }
        }
        channels 8
        format S32_LE
        rate 48000
    }
    ttable.0.0 1
    ttable.1.1 1
    ttable.0.2 1
    ttable.1.3 1
    ttable.0.4 1
    ttable.1.5 1
    ttable.0.6 1
    ttable.1.7 1
}

sudo nano ~/.asoundrc

pcm.!default {
	type asym
	playback.pcm {
		type plug
		slave.pcm "output"
	}
	capture.pcm {
		type plug
		slave.pcm "input"
	}
}

pcm.output {
	type hw
	card 1
}

ctl.!default {
	type hw
	card 1
}