shivasiddharth/GassistPi #391: Feature request : neopixel indicator

VH-Tech, 2018-06-14:
hi,
I wanted a feature in which 

1) If hot word is detected and the google assistant starts listening then there should be a certain animation with the neopixel

2) if  the request is being processed then also their should be an animation

shivasiddharth, 2018-06-14:
Already exists. Please go through the readme..

closed 2018-06-14 by shivasiddharth

VH-Tech, 2018-06-14:
I know that this feature already exists but it's for arduino  and I am requesting for Raspberry pi

shivasiddharth, 2018-06-14:
Without Arduino if you use the neopixels, the onboard audio jack on pi will not work properly. Both use pwm.

VH-Tech, 2018-06-16:
ok

cbinde, 2018-09-12:
For users without the need of the internal jack (e.g i am using usb sound device), it would be great to use the Neopixel directly connected to the Pi. Is there a chance to get that feature?