shivasiddharth/GassistPi #399: Just 16th Build a few errors and will not startup.

computerfreek274, 2018-06-17:
Let me start out by saying that GassistPI is Great I love it I have installed it on a few devices and it works great up to today. I do not know what is going on in this update. But I am sure it will be fixed. I am posting this just so you have a heads up on it. I have a copy on the last build so not a big deal. However I will hold off and test the next build for any errors to try and help fix this for others. I am not a Noob and understand what is going on it looks to be just some typos. No big deal.

Thanks for the great work.

Test working fine after I authorize device. Then I ran and installed startup service and then start the service but it never starts I reboot and no google.
Make the service installer executable

sudo chmod +x /home/pi/GassistPi/scripts/service-installer.sh
Run the service installer
sudo /home/pi/GassistPi/scripts/service-installer.sh    
Enable the services - Pi3 and Armv7 users, enable the "gassistpi-ok-google.service" and Pi Zero, Pi A and 
sudo systemctl enable gassistpi-ok-google.service    
Start the service - Pi3 and Armv7 users, start the "gassistpi-ok-google.service" and Pi Zero, Pi A and Pi 1 
sudo systemctl start gassistpi-ok-google.service  

So I moved on to try and start it   MANUALLY START THE ASSISTANT Replacing with my ID's
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'replace this with the model id'

I run it and I get this message so I take it this is why auto start will not work also.
 File "/home/pi/GassistPi/src/main.py", line 83
    kodi = Kodi(kodiurl, ,configuration['Kodi']['username'], configuration['Kodi']['password'])
                         ^
SyntaxError: invalid syntax

Also on a side note 
on line 19 of GassistPi-pip-requirements.txt the line was.
git+https://github.com/plamere/spotipy.git --update

i had to replace it with git+https://github.com/plamere/spotipy.git.

shivasiddharth, 2018-06-17:
> kodi = Kodi(kodiurl, ,configuration['Kodi']['username'], configuration['Kodi']['password'])

There are two commas between `kodiurl and configuration` i have fixed that in the all the scripts. try re-pulling the git.

computerfreek274, 2018-06-17:
Cool thanks. I will try tommorow. Did you also get around to the --update
thing at the bottom of my post?

Thanks great work.

shivasiddharth, 2018-06-17:
You need to add import tank to the list of imports. I have fixed it now thanks for pointing it out.

computerfreek274, 2018-06-17:
This build is not easy 123 like the last build then I did not need to install anything extra or import. I got past the yaml by doing source env/bin/activate pip install PyYAML . Now a new set or errors. this is a clean updated flashed SD card. The last build I am able to just run your code and be off and going. Not sure what is going on with this update. Thanks for the help.Hope this helps you also in making this work out of the box like the last build.

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 34, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 50, in <module>
    spotify_token = credentials.get_access_token()
  File "/home/pi/env/lib/python3.5/site-packages/spotipy/oauth2.py", line 63, in get_access_token
    token_info = self._request_access_token()
  File "/home/pi/env/lib/python3.5/site-packages/spotipy/oauth2.py", line 77, in _request_access_token
    raise SpotifyOauthError(response.reason)
spotipy.oauth2.SpotifyOauthError: Bad Request

bbbf0621, 2018-06-17:
i did the same steps like you did to manually launch this project too, but i'm unable to play Spotify. When ever i tried to stream music from Spotify it crashes.

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play song from Spotify"}
stopping vlc
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
song
1
song
Smart Baby Lullaby Row, Row, Row Your Boat
pi@raspberrypi:~ $ 

i have a playlist called song in Spotify, anyone knows any idea why?

aandroide, 2018-06-17:
@bbbf0621 
spotify does not work for me either. youtube works for me, have you copied the youtube apykey in the config.yaml file?

bbbf0621, 2018-06-17:
@aandroide yes i have key in the config file, but it still goes back to the terminal

Diastasis, 2018-06-17:
First of all, thank you very much for your work and effort.
After the installation on my Pi3 and device authorization the test worked fine. Next, auto-start installer executed.   However, after the reboot there is no reaction. Even if I try to start the assistant manually there is the following error:

/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'xxx-xxx-xxx-xxx'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 34, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 15, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 9, in <module>
    from youtube_search_engine import youtube_search
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 5, in <module>
    import yaml
ImportError: No module named 'yaml'

Any advice?

bbbf0621, 2018-06-17:
@DismissedGuy  try install PyYAML as computerfreek274 did.

Diastasis, 2018-06-17:
Solved. Thank you guys!

aandroide, 2018-06-19:
take away from me a doubt please. but in the spotify credentials, what did you put as a nickname? the name of the app you have created?

bbbf0621, 2018-06-21:
@aandroide  yes, i have spotifiy user name put in the config file too. I just found out recently GassisPi not only crash to spotify, but also when Google try to play news , radio too. It seems to be it crashed whenever it tries to play external audio.

commit 03c42eda (2018-06-21): Update GassistPi-pip-requirements.txt

Fixes #399 fixes #402
closed 2018-06-21 by shivasiddharth