shivasiddharth/GassistPi #404: PushButton with Hotword

cypricus, 2018-06-18:
Hi,
There is any way to active a relay or use pushbutton with an active command?
Only in this way i have "pt-BR", portuguese available to use.

I'm doing to my childrens use (Thay can't now speak english well)
Like a good father i'm trying here.

There is any way to use a hotword "Ok google" or anything that make the click button interaction?
In this way pushbutton the portuguese work like a champ.

Thanks a lot!

abovedmc, 2018-06-18:
I have the same question about using a hot word rather than a push button to activate the assistant.  I’m still stuck on getting GPIO 17 to become the push button.

cypricus, 2018-06-19:
Use a relay for that!
In a command trigger a relay to on/off the GPIO 17.
;D
Then send me the code too! lol

cypricus, 2018-06-19:
In pushbutton.py GPIO 22 and GND it's the trigger to activate "Ok google"
and start listening.
You can use a external trigger for that and this will just activate a relay
to click.

I can test with a Arduino for that with a word "Hello Duino", so will
trigger the relay and the GPIO 22 with GND on Raspberry.
But off course put this in the Rasp directly it's much better.

Forget about trigger the GPIO 17 now. Use a command word now with
pushbutton only to active relay and begin listen.

;D

2018-06-19 15:32 GMT-03:00 Carlos Arnt <cypricus2099@gmail.com>:

shivasiddharth, 2018-07-28:
I have added activation by voice/custom wakewords for even Pi Zero boards. Please check the last commint/update.

closed 2018-07-28 by shivasiddharth

jbrepogmailcom, 2018-12-12:
Hello! The install script on Raspberry Zero W says "Your board does not support Ok-Google Hotword". How can I use hotword please? Thank you

shivasiddharth, 2018-12-12:
@jbrepogmailcom  You have just posted one half of the message. The other half that you have not posted is the answer to your query.