shivasiddharth/GassistPi #407: Error in line 56 gassist-installer-pi-zero.sh

romeok01, 2018-06-20:
After instalation i have error in line 56 in gassist-installer-pi-zero.sh

Testing the installed google assistant. Make a note of the generated Device-Id
/home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh: linia 56:  9029 Błędna instrukcja     googlesamples-assistant-pushtotalk --project-id $projid --device-model-id $modelid

When i run manualy i have error:

pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id voice1 --device-model-id voice1-25123-pi-n03y3g
Błędna instrukcja

romeok01, 2018-06-20:
Now i have error:

/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id voice1-xxxx
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 34, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 15, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 9, in <module>
    from youtube_search_engine import youtube_search
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 8, in <module>
    import yaml
ImportError: No module named 'yaml'

k15866, 2018-06-21:
I too encountered this error. Initially i tried installing pyyaml as per google, with

sudo pip install pyyaml

which did not change anything. 

after uninstalling pyyaml in pip and pip3 with

sudo pip uninstall pyyaml
and
sudo pip3 uninstall pyyaml

and reinstalling just pip3 I have got rid of the bottom 2 errors

File "/home/pi/GassistPi/src/mediaplayer.py", line 9, in
from youtube_search_engine import youtube_search
File "/home/pi/GassistPi/src/youtube_search_engine.py", line 8, in
import yaml
ImportError: No module named 'yaml'

Now only getting first two with
ImportError: No module named 'mediaplayer'

Still in progress trying to find how to correct this.

k15866, 2018-06-21:
have tried reinstalling pyyaml with pip and pip3 without success

shivasiddharth, 2018-06-21:
Duplicate of #399

closed 2018-06-21 by shivasiddharth

romeok01, 2018-06-21:
I install pyyaml:

source env/bin/activate
pip install pyyaml

Now i have error:
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id voice1-xxxxx

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 34, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 50, in <module>
    spotify_token = credentials.get_access_token()
  File "/home/pi/env/lib/python3.5/site-packages/spotipy/oauth2.py", line 63, in get_access_token
    token_info = self._request_access_token()
  File "/home/pi/env/lib/python3.5/site-packages/spotipy/oauth2.py", line 77, in _request_access_token
    raise SpotifyOauthError(response.reason)
spotipy.oauth2.SpotifyOauthError: Bad Request

ankurshah963, 2018-06-25:
i have error

previous version install in my rpi2 and work complete fine. 
but new release version not install

i will follow all steps but last 
i will enterd my project id correct but
**

=========== pls saw my error ============

Registering...Traceback (most recent call last):
File "/home/pi/env/bin/googlesamples-assistant-hotword", line 11, in 
sys.exit(main())
File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/library/hotword.py", line 133, in main
device_model_id, device_id)
File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/device_helpers.py", line 76, in register_device
raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device PERMISSION_DENIED (403): Could not create the device instance. You don't have permission to get device model gapi. Make sure you registered this device model under your project gapi-195009.

i hope reply me and resolve that

thank you
ankur shah