shivasiddharth/GassistPi #409: When finished in Pi Zero error line 56

romeok01, 2018-06-21:
I install on Raspberry Pi Zero and finished i have error:

Testing the installed google assistant. Make a note of the generated Device-Id
/home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh: linia 56:  1500 Błędna instrukcja     googlesamples-assistant-pushtotalk --project-id $projid --device-model-id $modelid

Shidiqadm, 2018-06-21:
I also faced the same issue but in rpi3, I resolved by reinstalling it

romeok01, 2018-06-21:
Thanks i reinstall and when i run manually i have error:

 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id voice1-25123-pi-n03y3g
[01834da0] pulse audio output error: PulseAudio server connection failure: Connection refused
[0186cdf0] pulse audio output error: PulseAudio server connection failure: Connection refused
Błędna instrukcja

shivasiddharth, 2018-06-22:
main.py is not for Pi Zero. Please go through the Readme document carefully.

closed 2018-06-22 by shivasiddharth

ankurshah963, 2018-06-25:
I also faced the same issue.

how to the generated Device-Id??