shivasiddharth/GassistPi #410: Youtube and Spotify error

Glider95, 2018-06-22:
Hello Shivasiddharth !

I managed to install GassistPi on my Raspberry Pi 3 !
But now that I want to play either Youtube or Spotify music, I get an error.

I added API Key for Youtube in config.yaml and also ident for Spotify .

But now when I say '' Ok  Google, joue ma playlist All sur Spotify'' ( Hey Google, play my playlist All on Spotify) I get as return from Google >
'' Une erreur s'est produite. Réessayez dès que vous pouvez.'' = '' An error occured . Try again later '' 
The assistant is NOT crashing.
It's the same problem for Spotify and Youtube !

`#Spotify credentials. Register for a spotify developer and get the credentials.
Spotify:
  client_id:2a7d834997d446*******************
  client_secret:87c4304ea5034860a3**************************
  username:11709***`

`API Key for YouTube and Kickstarter Search Engine
Google_cloud_api_key:AIzaSyA0yUQSw8_bRW6AGZL***********************************`

closed 2018-06-22 by shivasiddharth

Glider95, 2018-06-22:
I just update actions.py and I still have same error :/

I think it's not from APi declaration, I must have something wrong somewhere else

shivasiddharth, 2018-06-23:
The search for customizations have been made in english. If you are giving commands in other languages, the you obviously need to change the customization language.

Glider95, 2018-06-23:
Ok ! So either I switch my assistant to English from my phone ?
Either I modify your script to change customization word in French ? ( Just need to write them in French ?)

commit db32fb1f (2018-06-22): Update actions.py

Fixes #406 Fixes #410